prusa3d/PrusaSlicer #13302: V2.8.0 - Print Settings - Output Options - Post-processing scripts. Edit box obscured and not selectable or editable.

philjynx, 2024-08-26:
### Description of the bug

[image]

I've just upgraded to this version of PrusaSlicer.

The project still has and still executes my post processing script but I cannot edit the reference to it.

The same applies to a new project so, one is unable to use post processing unless you have the script already referenced in a print settings definition.

### Project file & How to reproduce

[BridgeFlowRatioTest.zip](https://github.com/user-attachments/files/16752473/BridgeFlowRatioTest.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

V2.8.0

### Operating system

Windows 10 latest version

### Printer model

A home brewed knock off of an ultimaker!

philjynx, 2024-08-26:
I will revert to the older version until this issue is resolved - as an aside it also means I won't have that annoying nag about scripts every time I open a project.

neophyl, 2024-08-26:
Duplicate issue.
https://github.com/prusa3d/PrusaSlicer/issues/12932
https://github.com/prusa3d/PrusaSlicer/issues/12925

philjynx, 2024-08-26:
That being the case, a decent way of searching the issues is needed. I used
the search facility that was offered and there was no match, hence my
apparent duplication.

lukasmatena, 2024-08-28:
Closing as a duplicate of #12932. It will be fixed in 2.8.1, which will be released soon.

closed 2024-08-28 by lukasmatena