prusa3d/PrusaSlicer #13308: Adding IDs to object names when using klipper exclude object labeling

HelgeKeck, 2024-08-28:
### Description of the bug

Klipper excludes objects based on their names and not based on IDs like octoprint does.  
[image]

When having multiple objects with the same name then PS creates this here:
[image]

In this case the klipper exclude object feature cant differentiate anymore between the obejcts and cancels all of them when the user just want to cancel one obejct.

It would be nice if PS could add the object ID to the end of the name object to fix this issue

### Project file & How to reproduce

[Shape-Box_0.2mm_PETG_1m.zip](https://github.com/user-attachments/files/16784740/Shape-Box_0.2mm_PETG_1m.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 10

### Printer model

RatRig V-Minion

eduncan911, 2025-04-13:
note that this was also fixed in the PR #13641

HelgeKeck, 2025-04-14:
nice, then i close this here

closed 2025-04-14 by HelgeKeck

eduncan911, 2025-04-18:
it's not merged yet.  lol

HelgeKeck, 2025-04-18:
> it's not merged yet. lol

this is jsut a bug report without any PR code, so its fine if i clsoe this one here