prusa3d/PrusaSlicer #13316: All Filament settings are gone after 2.8.0 update

markusdd, 2024-08-31:
### Description of the bug

see title
[image]

I'm furious. All my printer profile and filament settings are completely gone after updating to 2.8.0. Trying to restore from a config snapshot I took before update says they are not compatible.

How do I get these back? This is a catstrophe, all of my projects are broken because of this?! 
Did no one even test this update?!

### Project file & How to reproduce

update to 2.8 from 2.7.4

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+win64

### Operating system

WIn11

### Printer model

Elegoo Neptune 3 Pro

markusdd, 2024-08-31:
Ok. I finally figured out what happened. After hours of searching.

The new config rollout forced me into installing the updated elegoo profiles.

These lead to the fact that my existing printer profile is now not marked as derived from the Neptune 3 Pro base profile, but rather from 'default'.

This has the ripple on effect that ALL print profiles and filament profiles now are incomptible with this printer and hence none of them are displayed, although they are there.

By manually merging the settings I wanna keep into the new profile, saving to an extra name, and then saving back to my existing profile, I can make at least the print settings and filaments reappear. UNfortunately that still bricks all old .3mf projects that had project specific modifications applied to printer, filament and/or print settings.
These are now no longer considered the same base and you are greeted by this mess:
[image]

So this 0.2mm profile which is exactly the same in the new version had modifications, and I cannot even find out which because it is considered to be only compatible with another base and hence it won't even let me compare what the diff was:
[image]

This is completely messed up.
The condifgs are literally the same, why does 2.8.0 force me to update and destroy all the settings and projects I built over the last 2 years?

This literally makes me mad.

lukasmatena, 2024-09-03:
You have already posted this in #13205, where @kocikdav offered assistance and asked you to provide some input to let us check what exactly happened. You have closed that issue without an answer.

The migration from 2.7.x to 2.8.0 was tested thoroughly, and since you are apparently the only person suffering from this issue, it indicates that it might be something specific to your setup. But without having a look at the actual contents of you config folder, it is impossible to guess.

markusdd, 2024-09-03:
You stating that it's only me already suggests you haven't even tried to look at it.

Don't care anymore, I've switched Slicers.

closed 2024-09-03 by markusdd