prusa3d/PrusaSlicer #13320: (Prusament) ASA cannot be selected when using MMU3 on MK4

ErGeeBee, 2024-09-01:
### Description of the bug

In the configuration wizard, it is not possible to select ASA filament for the MK3.9, MK3.9S, MK4, MK4S  & MMU3 combination. 

For the MK3.5/MMU3 it is possible.

Therefore I cannot print ASA without disabling the MMU3 on my MK4.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

macOS Sonoma

### Printer model

MK4 + MMU3

techis, 2024-09-11:
I also reported this here: https://github.com/prusa3d/PrusaSlicer/issues/13260 Greatly annoying to upgrade the printer and lose features.

rtyr, 2025-02-06:
MMU3 (MK4/MK3.9) support for selected ABS, ASA, BVOH and PC blend filaments was just added via configuration update 2.1.10. Until this update, only PLA and PETG filaments were officially supported.

closed 2025-02-06 by rtyr