prusa3d/PrusaSlicer #13322: ABS Filament not an option

phoov1, 2024-09-02:
### Description of the bug

In the configuration wizard, under filaments, all filaments are selected.  When I try to select filaments for the MMU3, ABS filaments other than PLA and PETG are not listed.  I printed ABS a few days ago with no problems.

[image]

### Project file & How to reproduce

[trd_toyota_oem_style_trailer_hitch_plug.zip](https://github.com/user-attachments/files/16831075/trd_toyota_oem_style_trailer_hitch_plug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 10 pro, Version 22H2, OS Build 19045.4780, Windows Feature Experience Pack 1000.19060.1000.0

### Printer model

MK4 upgraded from a MK3S

neophyl, 2024-09-02:
I think this is the same issue as https://github.com/prusa3d/PrusaSlicer/issues/13320

techis, 2024-09-11:
Same for ASA. I also reported this here: https://github.com/prusa3d/PrusaSlicer/issues/13260

rtyr, 2025-02-06:
MMU3 (MK4/MK3.9) support for selected ABS, ASA, BVOH and PC blend filaments was just added via configuration update 2.1.10. Until this update, only PLA and PETG filaments were officially supported.

closed 2025-02-06 by rtyr