prusa3d/PrusaSlicer #13325: post-processing script text box blank

ghylander, 2024-09-02:
### Description of the bug

in PS 2.8, the psot processing text box appears blank.

The text is actually present and it can be modified, but it's not visible:
[image]

If you type anything in the box, then switch print profile, you will get the changes dialog, showing the changes:
[image]

You can also click the box, then ctrl+A ctrl+C and it will copy all text to the clipboard

### Project file & How to reproduce

[Untitled.zip](https://github.com/user-attachments/files/16836576/Untitled.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

w11

### Printer model

ender3 s1 pro

neophyl, 2024-09-02:
Please search for duplicate issues.  This has been reported multiple times.  the current open one of these is https://github.com/prusa3d/PrusaSlicer/issues/12932

As mentioned in one of the closed duplicates here https://github.com/prusa3d/PrusaSlicer/issues/12925#issuecomment-2315537129  it will be fixed in 2.8.1 which apparently is not too far off now.

ghylander, 2024-09-02:
I did search a bit, but there's many unrelated issues and the one you posted is dated 27 of June.

I apologize for not finding that issue in my manual review of the 1000s of currently open issues.

There's also no "known issues" section, could greatly help avoid these duplicates.

closed 2024-09-02 by ghylander