prusa3d/PrusaSlicer #13327: Command line in shortcut to select printer configuration doesn't work

srjacob999, 2024-09-02:
### Description of the bug

I am trying to create Windows shortcuts for my Prusa XL and MK4.  I tried putting the configuration profile on the command line in the shortcut like:

"C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe" --printer-profile XL

But Prusaslicer doesn't load.

### Project file & How to reproduce

Add the line : "C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe" --printer-profile XL

to the shortcut that Prusaslicer creates.  Make sure you have a profile  labeled XL.

Then try clicking on the new shortcut.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windoes 10

### Printer model

Prusa XL

lukasmatena, 2024-09-04:
Thanks for the report. Running the command from the command line using the `prusa-slicer-console` will result in this:
```
C:\Users\lukas\Downloads\private\slicers\PrusaSlicer-2.8.1-rc0+win64-202408290855>prusa-slicer-console.exe --printer-profile XL
Request is not completed. All of Print/Material/Printer profiles have to be entered
```

The three options have to be all present. However, even if you added the two remaining ones, you would not get the behavior that you expect. The `--printer-profile`, `--print-profile` and `--material-profile` command line options were only meant for command line slicing. You are clearly using them with the assumption that they will start the GUI slicer and select the requested profiles, which is not supported and PrusaSlicer should actually report an error instead of just starting GUI and ignoring the parameters silently.

I admit that what you are trying to achieve makes sense. We will add an error report for now, and later likely extend the functionality to what you wanted to do.

Did I understand the issue correctly?

srjacob999, 2024-09-04:
You understood the issue correctly.  I would like an icon on my Windows display to select a profile for a particular printer.  I have an MK4 and an XL, and would like to be able to select the slicing profile based on which icon I click on.

neophyl, 2024-09-04:
In the meantime couldn't you create and save 3mf projects for the configurations you want and then if you double click on one of those configure it to start PS (make PS the default action for a 3mf).  That would launch PS and open the project.  As the project would contain the 3 profiles selected when saved doesn't that fulfil what you want in practice ?

srjacob999, 2024-09-04:
The problem is that when I save a configuration, it goes into a big file, so that won't work.  If you could save each configuration in a separate file, your suggestion would do what I want.

srjacob999, 2024-09-04:
Tried it.  It doesn't even launch Prusaslicer.

neophyl, 2024-09-05:
I don't know what you mean about a big file.  I'm not talking about saving a configuration.  I'm talking about a normal Prusa Slicer Project File.  File>Save Project As.  This should only be a few k.   With minimal objects like a cube mine work out to about 9K.  Hardly a large file.

Open PS, select the 3 profiles you want to use, add a simple cube and make it non printable and move it off the plater.  Then Save Project.  Say MK4_Default as an example.  Then change the 3 profiles to the next configuration you want.  Say an XL,  Save that one as XL_Default.  

Then when you open the project file (a 3mf type) it opens with your selected profiles (and any other changes you may have at the time).  

The fact that double clicking on a .3mf file doesn't open Prusa Slicer is a different problem.  That can be configured in your operating system by changing the default program used for 3mf.  A simple google search on how to do that for your chosen OS can detail how.

srjacob999, 2024-09-05:
I trid all the things you suggest, including changing the default in Windows to have Windows default to 3mf file to open Prusaslicer.  When I double click on the 3mf file, nothing happens.  Prusaslicer doesn't get launched.  I checked the Windows 3mf default, and it definetly points to Prusaslicer.

neophyl, 2024-09-05:
I currently use this method on 3 different windows machines, an older windows 8 machine and 2 windows 10 machines.  It has worked as a method all the way back from the earliest version of Prusa Slicer up to 2.8 the current release.  Double clicking a file type and it launching a specific program is a Operating system function.  So if its not launching PS then I would be searching windows forums looking for an answer.
What windows version are you running ?

Out of curiosity, what happens if you configure windows to open with a different program such as 3D Viewer ? and then double click a 3mf file.

Also just had a thought, is the version of Prusa Slicer in the windows default the CURRENT version of PS or an older version ?

srjacob999, 2024-09-05:
Forgot to mention.  Prusaslicer is the current version (2.8)

lukasmatena, 2024-11-21:
Implemented in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena