prusa3d/PrusaSlicer #13328: Add "only retract when crossing perimeters" to the list of settings you can change per object

thatguymendel, 2024-09-02:
When the nozzle travels through thin walls with "only retract when crossing perimeters" enabled, the nozzle can leak filament onto the print which affects the surface quality of the print.
   If i wanted to print a model that has thin walls together with other models that don't, i'd have to turn off "only retract when crossing perimeters". 
   Obviously you can just print that model separately with that setting disabled but it would be nice to be able to print it together with other models having that setting enabled. 

The solution would be to allow you to enable/disable this setting using the object settings.

[image]

[image]

The solution would be to allow you to enable/disable this setting using the object settings.

u89djt, 2024-09-03:
(fellow user who sees something in your photo that is different to the problem you describe) If you can fill out the rest of the bug report according to the instructions, maybe someone can help you with what's happening there. Make sure it's exactly the 3mf file used to generate the print. You need to zip the 3mf file.

thatguymendel, 2024-09-04:
Could you describe what you think is the problem? I have printed the exact same file with only retract when crossing perimeters disabled and it prints perfectly. This setting stops any retract from happening and if your extruder is prone to leaking it will leak into the print. This is not a problem when for example you are printing a cube where the nozzle doesn't get close to the perimeter when traveling. In the model i was trying to print the the walls are 1mm thick which with a 0.4mm nozzle will be printed with 2 outside perimeters and 1 small inside perimeter. I forgot to mention that i have "avoid crossing perimeters" enable which is causing the printer to travel along the thin wall. Having both these settings enabled at the same time means the when the printer is traveling along the wall the nozzle leaks filament onto it and because the wall is thin you can visibly see this on the outside surface. Disabling either of these settings would solve this issue, but right not you are not able to disable either for just one object
[image]

u89djt, 2024-09-04:
Don't forget the 3mf files. The right thing to do is to post both; the one for the top example, and the one for the bottom example. That way everyone's absolutely certain what's going on. Save the project as sliced and zip that.

u89djt, 2024-09-04:
And state operating system, slicer version and which printer you're using. The 3mf files will include that information, but type it out. Make it easy for extremely busy staff to see what's going on. This is all about gathering knowledge from people who know stuff you don't know, so follow their instructions to the letter when you can.

thatguymendel, 2024-09-04:
[things.zip](https://github.com/user-attachments/files/16873187/things.zip)
PrusaSlicer 2.7.4
sidewinder x1
Windows 10

u89djt, 2024-09-04:
This is what the preview looks like when your 3mf files are sliced in 2.8.0 with travel, de/retractions and seams views active. I'm not sure what you're viewing in your screenshot because you've cropped it. Did you get the files the other way round? The bottom file seems to have the travel that corresponds to the problem you describe.
You've ticked the avoid crossing perimeters box, but you've left the maximum detour at zero, so it's actually not been enabled.
[image]
OK, it seems like the due dilligence is done. Opinion: this random user thinks it'll be worth drying the filament more. If you haven't the time/space/equipment to dry the filament, there's probably some kind of workaround achievable. If you'd like some ideas for that, there's a bunch of folk watching the hub, and I'm sure one of us will chip in. Just make it known that you want help.
[image]

u89djt, 2024-09-04:
FWIW, per object retraction achieving what you hope for would need interrogation of the identities of objects the travel path crosses. That would be theoretically possible of course, but I wonder if that would rise to the top of a to do list.
There's a kind of lame question to ask, too: have you tried increasing retraction? I don't like that either :)

thatguymendel, 2024-09-04:
The paths look similar because avoid crossing perimeters is in fact enabled for both. The max detour length option limits how long a detour can be. Setting this option to 10mm will disable avoid crossing perimeters if the detour would be longer than 10mm. Setting it to 0 will disable "avoid crossing perimeters - max detour length" and it will always avoid perimeters regardless of the detour length. 
The difference between those two files is that one has only retract when crossing perimeters enabled and the other one has it disabled. That means when it travels along the path one will retract and z hop before the travel and the other will do neither.

The leaking filament is a different issue. My printer has a volcano nozzle which is more prone to leaking. Drying the filament may help, but I've tried it before and it didn't seem to do much.
Yes I have tried tuning retraction.

The problem here is that when it travels through the object there is no retraction.
Avoid crossing perimeters, only retract when crossing perimeters, and z hop are my workarounds for a leaking nozzle
-Avoid crossing perimeters
 this is useful for objects that have holes in them. The nozzle will go around the holes and not drag stringing through them.
-Only retract when crossing perimeters
 this is important because with z hop enabled, avoid crossing perimeters does nothing because the stringing stays connected to the nozzle until it lowers onto the print. 
-z hop
 . Without this after a long travel between objects, the nozzle will leak filament and it will get deposited on the outside surface of the print

u89djt, 2024-09-04:
Extremely sloppy of me not to perform the sanity check of trying switching off avoid crossing perimeters.
[image]

closed 2026-09-29 by prusaslicer-bot[bot]