prusa3d/PrusaSlicer #13332: PROBLEM NUMBER 6 NOT WORKING IN THE PRUSA SLICER MY KEYBOARD AND PC IST GOOD

jerson205, 2024-09-03:
### Description of the bug

NUMBER 6 NOT WORK IN PRUSA SLICER
[image]

### Project file & How to reproduce

ANYWHERE 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8 WINDOWS 64

### Operating system

VERSION 2.8

### Printer model

ENDER 3 S1 PRO

fe60, 2024-09-04:
Sounds like a duplicate of #13043
One user complains about not being able to type a [4 from the num-row](https://github.com/prusa3d/PrusaSlicer/issues/13043#issuecomment-2295526189).

Do you have a numpad? Using it could be a workaround until this problem has been fixed.

jerson205, 2024-09-04:
Sorry PC portable 👨🏻‍💻

lukasmatena, 2024-09-12:
Closing as a likely duplicate of #13043, which is fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases).

closed 2024-09-12 by lukasmatena