prusa3d/PrusaSlicer #13339: Infil inconsistent layers around a hole.

VH-fixer, 2024-09-06:
### Description of the bug

 Infil is done inconsistently around a hole or circular profile.   It ends up being an unsupported strand on one or more layers.   
See the arrows in the pictures.    
[image]

### Project file & How to reproduce

[erratic-infil.zip](https://github.com/user-attachments/files/16909399/erratic-infil.zip)
This is the project that will reproduce the problem.  

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

Windows 11 23H2

### Printer model

MK4 with MMU3

neophyl, 2024-09-06:
That is controlled by your infill anchor settings.  If you change those it slices differently so is already under user control

[image]

[image]

u89djt, 2024-09-06:
(fellow user with good news) Hi there, that infill as absolutely fine: it's adhered to the cylinder in the same way infill anchors adhere everywhere else, including in the rest of your object there. The changes in location are due to the change in shape of the object as it rises. Try rocking the view up and down a bit so you can see the relationship better. Here it is from a different perspective: 
[image]

closed 2026-09-29 by prusaslicer-bot[bot]