prusa3d/PrusaSlicer #13342: Command-Q does not work in PrusaSlicer when on the PrusaConnect tab

ChinookTx, 2024-09-07:
### Description of the bug

Normally, when pressing Command-Q in an app on a Mac, this quits/closes the app.  The shortcut works on Plater/Print Settings, etc tabs, but not on the Prusa Connect tab.

### Project file & How to reproduce

Happens with any project/settings.  Just navigate to the Prusa Connect tab and try to quit the app with Command-Q.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0+MacOS-arm64

### Operating system

MacOS 15 24A5320a

### Printer model

MK4S

ChrisHeerschap, 2024-10-11:
Didn't see this when I opened up a duplicate ticket, but want to confirm that I'm seeing the same with 2.8.1 as well.

A-Printer, 2024-10-15:
The Command-Q keyboard shortcut is not the only one that is disabled when in this tab. Command-H (to hide the app) and all shortcuts from the Window menu (allowing us to switch between tabs) also can't be invoked.

ChrisHeerschap, 2024-10-17:
Hadn't tried that, but can confirm that CMD-H doesn't work either. Would also love a keyboard shortcut for that tab, but man I'd love if CMD-R would actually reload the page when it goes blank white. (different issue still outstanding)

closed 2026-09-29 by prusaslicer-bot[bot]