prusa3d/PrusaSlicer #13343: Add "Infill Line Multiplier" (an option exposed by Cura)

crpalmer, 2024-09-07:
**Is your feature request related to a problem? Please describe.**

Lightning infill for large objects does not work well.  See for example issue #10442.

If you print a a 500mm in diameter cylinder using Lightning infill, it will create very long "bolts" and since each bolt is only a single line width they have very little change of not wobbling / collapsing and in many other ways failing.  Being able to increase the number of infill lines is one technique that can be used to make it possible to use Lightning infill for large volumes.

Right now I would not recommend using Lightning infill 

**Describe the solution you'd like**
Implement: "Convert each infill line to this many lines.  The extra lines do not cross over each other.  This makes the infill stiffer..."

This could be implemented in some generic fashion that would apply to all infills or it could simply be a Lightning infill specific option.   Issue #10585 is related (asking for the same functionality but looking more specifically for grid infill).  This issue is definitely geared more toward how Lightning infill works as it doesn't have the regular structure supporting it that the other infill types have.

**Describe how it would work**
See "describe the solution".

**Describe alternatives you've considered**
There are no options available at this point to control the Lightning infill.

**Additional context**
Currently using version  2.8.0.

See the attached photo.  In this print all of the infill around the outer perimeter looks great but for the "bolts" which have grown too long, all have failed and are building up a mass of spaghetti that I'm praying will magically save the print.  This example is more complex than just a cylinder in that it is building up extra long bolts to support several islands (the islands are the holes of dowel connectors for joining a split object).

Have the ability to make each "bolt" thicker would make this less likely to happen and therefore make it possible to use infill for large format printers that are printing large volumes without having to use an insane amount of plastic.

[image]

u89djt, 2024-09-08:
(intrigued fellow user) Can you post the 3mf file of your print there? I want to look at what happens if you set the infill extrusion to 0.6 in advanced settings and slow down the infill speed to, say, 100. Half speed single extrusions would take as long as two full speed extrusions, and a thicker extrusion won't take more than two. Something like this: 
[Shape-Cylinder.zip](https://github.com/user-attachments/files/16922209/Shape-Cylinder.zip)
[image]

crpalmer, 2024-09-08:
[dave middle section.zip](https://github.com/user-attachments/files/16922385/dave.middle.section.zip)

@u89djt This is more or less what I printed (I hadn't saved the original file but this is close enough).

I'm thinking along the same lines as you for my next print.  Last time, I was running the infill at 120mm/sec and a extrusion width of 0.65mm for a 0.6mm nozzle.  I'm going to change that down to 50mm/sec to give it more cooling time and bump the extrusion width up to 0.9mm.  This was the best I could come up with approximate the multiple lines functionality.  It appears to be working without issues on a small test print.  I have no idea how it will work for a large print...

rtyr, 2024-09-09:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7044. You issue will be linked.

closed 2024-09-09 by rtyr

u89djt, 2024-09-09:
@rtyr the proposed solution is the same, but the issues that require a different approach are unique to lightning? No other infill reaches out into empty space on its own. Infill back when #7044 was registered was all wall-to-wall. Lightning was introduced in Dec 2021, after that issue. Perhaps having an issue open that speaks to vulnerability of lightning infill in larger object could fill two functions: an opportunity for users with this problem to find an issue that says what they want to say, and to act as a focus for registering progress on long lightning.

crpalmer, 2024-09-09:
@rtyr Didn't I explicitly metion #10585 which is a duplicate of #7044 and say that this was different citing the same reasoning as @u89djt.  Lightning infill is very different because it doesn't have a full support structure below it ("not wall to wall") and implementing something specific for it could be a sensible approach.

Furthermore, the other reference bugs are asking for specifically grid infill.  Based on the focus of the request on grid infill, I suspect it will never get implemented and therefore lightning infill will evermore but only effective for objects up to a certain size. 

That is a shame.

u89djt, 2024-09-09:
@crpalmer there are other things that can be changed about lightning to contribute to its stability. The current implementation places branching as high as possible in the print, and starts from a vertical line on the wall. Starting at V shapes and optimizing the branching differently could make a significant difference. Lots of possibillities.