prusa3d/PrusaSlicer #13348: Tab physical printer don't work

Siskin95, 2024-09-09:
### Description of the bug

Tab "physical printer" show:
" Connection failed"
Something went wrong.

### Project file & How to reproduce

Tab "physical printer" show:
" Connection failed"
Something went wrong.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Win11

### Printer model

MK3S+ with Raspberry zero 2W

Robagon3D, 2024-09-25:
+1
Version 2.8.1+win64
Win 11
MK3S+ with RaspPi zero 2W

JackpwMurray, 2024-09-25:
I have similer problem, except I can connect to my mini+, but the login menu fails to apear, rendering it useless. Can still access through IP though.

Jan-Soustruznik, 2024-10-08:
Hello, @Siskin95 m @Robagon3D , @JackpwMurray , Thank you for your feedback. Could you provide log according to steps below. 

1. open cmd.exe (command line)
2. go to Slicer folder (where the slicer is installed Program files > Prusa3D > PrusaSlicer)
3. find the "prusa-slicer-console.exe"
4. drag and drop the "prusa-slicer-console.exe" into the CMD window  - you will see that the path will be added
5. add to end of the path this command " --loglevel=5" and press ENTER
6. PrusaSlicer will start
7. go to physical printer tab in PrusaSlicer application - to evoke problem mention above
8. end the PrusaSlicer 
9. copy all text in CMD window (in the log may be Printer IP address, access key - you can delete it in log, or you can send me a directly into my email if you dont want to share this log public, my email  jan.soustruznik@prusa3d.cz - let me know that you sent me a email thru this ticket)

Siskin95, 2024-10-09:
Email with log was sent.

Robagon3D, 2024-10-18:
Email with log was sent.

closed 2026-09-29 by prusaslicer-bot[bot]