prusa3d/PrusaSlicer #13357: [BUG] Copy paste does not retain position of Part modifiers in the same order in the part tree.

thew1se, 2024-09-11:
### Description of the bug

While in the plater, when choosing a part+its modifiers, to be copy pasted, the pasted part does not always keep the position of the part and modifier in the tree in the same order.  As a result, the modifiers do not always effect the part as expected.

This occurs what appears randomly.  Cant figure it out.  If I delete the pasted part and do it a second time, it nearly always works properly.

Also, I could not drag the modifiers to its correct position in the tree, so had to redo the copy paste to make it work.

### Project file & How to reproduce

[Prusa slicer copy paste problem.zip](https://github.com/user-attachments/files/16970504/Prusa.slicer.copy.paste.problem.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 11

### Printer model

MK3s+

Jan-Soustruznik, 2024-10-07:
Hello, @thew1se , thank you for your report, but we cannot reproduce reported problem. Could you check again and provide some steps what exactly you did? Or make a video? Thank you.

thew1se, 2024-10-08:
Steps to recreate (had to test a bit so sorry it wasn't there the first time):

1. Replace STL with a similar STL but different size.
2. Move modifier to correct location
3. select entire model
4. Ctrl+C, Ctrl+V 

Video attached.

[Video of Problem.zip](https://github.com/user-attachments/files/17298966/Video.of.Problem.zip)

closed 2026-09-29 by prusaslicer-bot[bot]