prusa3d/PrusaSlicer #13361: PrusaSlicer-2.8.1-rc1+linux-x64-newer-distros-GTK3-202409121246.AppImage on Ubuntu 24.04 looking for libwebkit2gtk-4.0.so.37

ssteinerx, 2024-09-12:
### Description of the bug

On Ubuntu 24.04, AppImage won't start with:

    /tmp/.mount_PrusaSgfhysT/usr/bin/bin/prusa-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

libwebkit2gtk-4.1is installed on this version of Ubuntu. 4.0 doesn't even show up in an `apt search` except for the doc

### Project file & How to reproduce

Download, set +x attr, try to launch.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.1-rc1+linux-x64-newer-distros-GTK3-202409121246.AppImage

### Operating system

Ubuntu 24.04

### Printer model

printer

curvatura99, 2024-09-12:
Same issue here. I tried to to symlink libwebkit2gtk-4.0.so.37 to libwebkit2gtk-4.1.so, but that just led down the rabbit hole looking for other 4.0 dependencies.

lukasmatena, 2024-09-12:
Fixed. Sorry about that. The current two AppImage build requires too many manual steps... Closing.

closed 2024-09-12 by lukasmatena