prusa3d/PrusaSlicer #13371: Thank you 2.8.1-rc1

mrwizardtx, 2024-09-13:
Thank you for listening, 2.8.1-rc1 team.

A nice to have, would be the consideration of adding an import button menu on the fat menu, (plater,print,fil,printers) considering 90% of the tool is used for importing.

lukasmatena, 2024-09-13:
Hi. If we are talking about importing geometry (STL/3MF), the leftmost button in the top toolbar should do the trick.
[image]

Are you often importing something that you need the menu for?

dwprod38, 2024-09-17:
Same issue in post #13059 and #12908 : G-code preview performance is very bad with PrusaSlicer 2.8.1 rc1 when moving bed with mouse (refreshing every 2 to 4 secs) but in 3D editor view : everything move smoothly... Decide to downgrade to PrusaSlicer 2.7.4 

Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Sierra Version 10.12.6 (Build 16G2136)
Total RAM size [MB]: 8,590MB
OpenGL installation
GL version: 3.3.0 (3.3 NVIDIA-10.2.37 310.90.10.05b54)
Profile: Core
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce GT 330M OpenGL Engine
GLSL version: 3.30.0
Textures compression: Enabled
Mutisampling: Enabled (4 samples)

closed 2026-09-29 by prusaslicer-bot[bot]