prusa3d/PrusaSlicer #13374: Prusa Connet tab is still completely empty

Kaelum, 2024-09-13:
### Description of the bug

I just upgraded to `Version 2.8.1-rc1+win64` from the 2.8.0 release, and there is no difference in the `Prusa Connect` tab.

### Project file & How to reproduce

No project file, just open PrusaSlicer and click on the `Prusa Connect` tab.  Here is a screenshot of the tab:

[image]

This tab has never worked for me, so the issue reported in the `Change Log` does not apply to me, as it states that the issue only occurs after the tab has been opened for some time.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.1-rc1+win64

### Operating system

Windows 10

### Printer model

N/A

bkerler, 2024-09-14:
Same issue with 2.8.1 and linux (ubuntu 24.01) with latest 6.2.0-alpha1+8672 firmware and using appimage. No connect and no "send to printer" working anymore. Using 2.8.0, everything works as expected and printer can be seen.

bkerler, 2024-09-20:
[image]
When I log out of the Account, I can't even login back, login screen is empty.

bkerler, 2024-09-20:
Note: I used appimage. Using flatpak, I have the issues on both 2.8.0 and 2.8.1.

bkerler, 2024-09-20:
Double of issue #[#12969](https://github.com/prusa3d/PrusaSlicer/issues/12969).
Quickfix is: start with WEBKIT_DISABLE_COMPOSITING_MODE=1

Kaelum, 2024-09-20:
> Double of issue #[#12969](https://github.com/prusa3d/PrusaSlicer/issues/12969). Quickfix is: start with WEBKIT_DISABLE_COMPOSITING_MODE=1

The issues may be somewhat similar, but my observation of the `Prusa Connect` tab has never changed since it was first introduced, other than the new clickable icon in the bottom right.  The page has never displayed anything other than a blank white page, which would eventually change to Connection Failed message after some time.

closed 2026-09-29 by prusaslicer-bot[bot]