prusa3d/PrusaSlicer #13375: Add Additional "Slow" and "Super Fast" print profles

bryn51, 2024-09-13:
**Is your feature request related to a problem? Please describe.**
Prusa Slicer default Print Profiles include "Fast Detail", "Speed" and "Structural" profiles, missng "Slow" and "Super Fast" profiles

**Describe the solution you'd like**
Develop both Slow profiles and Super Fast Profiles for Prusa Slicer based on capabilities of MK4S printer, for use on slow printing material like TPU, and other materials that can now be printed faster using CHT Nozzles.  The profiles would be made compatible with MK4S/Mk4/XL printers having CHT nozzle as available option, with enhanced cooling.  Include a number of layer height options.

**Describe how it would work**
Develop and test the profiles using available expertise in Prusa Research. Distribute the additional profiles using standard PS profile distribution system.

**Describe alternatives you've considered**
Users have the capacity to develop their own profiles.  The distributed profiles are a starting point and the default for lazy folks (eg everyone).  This this is a convenience feature.

**Additional context**
There are folks out there not purchasing MK4S and Upgrades because PS is telling them that there is no speed benefit, as advertised in Prusa Marketing literature. Adding these additional convenience profiles is a relatively fast way to deliver the advertised speed improvements.

rtyr, 2024-09-14:
It is already like that. If you select HF nozzle, then you will get different variants of the profiles. The actual time difference largely depends on the type of object.

closed 2024-09-14 by rtyr

bryn51, 2024-09-15:
> It is already like that. If you select HF nozzle, then you will get different variants of the profiles. The actual time difference largely depends on the type of object.
> 
> As for more options and variants, we are not planning to add more at this moment.

You must have a different version of PS than I do.  I have 2.8.0, and printers installed are MK4 and MK4S.  There is no "slow": print profile available. We are talking 11-25 mm/s here.  As for "Very Fast" I spoke to Prusa Support today, and they indicated this was being worked on but not yet ready for prime time.
I looked at 0.25 mm Fast and 0.4 mm Fast, and both are a stately 120 mm/s, not what I'm expecting on the MK4S.   The 0.32 LH goes to 145 mm/s. 
What speed is that "bonkers" benchy again ?