prusa3d/PrusaSlicer #13376: Linux AppImage -> FLATPAK notes in 2.8.1-rc1

eduncan911, 2024-09-14:
A serious caution about the Flatpak note for Linux users [in the latest 2.8.1-rc1 release notes](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1).

By design, Flatpak purposely does not have access to the standard XDR config directories - where all PrusaSlicer config profiles live and are stored today.  E.g. `~/.config/PrusaSlicer`.  There are a few decisions to make when switching to Flatpak, and this issue is opened to start those discussions publicly so we can have a word in how this is developed.  :)  

Basically, we are hoping to avoid the ugly additional steps such as:

    flatpak run \
      --filesystem=xdg-config/:create \
      --filesystem=xdg-cache/:create \
      --filesystem=xdg-data/:create <pruaslicer-app-id> 

But to keep PrusaSlicer sandboxed, that may be an acceptable upgrade path to those with previous prusaslicer installs.  And one that should be noted in the release notes.

Other users may have some other gotchas, but that's the biggest I've ran into over the years.

buhman, 2024-09-15:
This could be best solved by implementing #12968 , keeping AppImage builds, and disabling webkit by default on AppImage builds.

CRCinAU, 2024-09-19:
Ya know, I've sang Prusa's praises for keeping AppImage around and making it a small, self-contained package that 'just works' everywhere.

It's be a shame if it came to relying on flatpak due to developer laziness. It pulls in entire ecosystems of stuff that's already on systems in package format.

Pitel, 2024-09-19:
I also thought that including WebKit was just a quick hack to integrate PrusaConnect (which I'm actively using). But the next step should be to implement the proper API, so we don't need this huge library just to upload a file and show some icons. From what I remember, when integrating OAuth login, you'd just need to temporarily run a local web server to receive the tokens.

novaplusplus, 2024-09-20:
My own 2cents... I decided to try the flatpak version after being disheartened by the news appimage was going away (appimage is my preferred way of getting and keeping software on linux, I like having a drive full of applications I can just load up anywhere without needing to worry about repositories and such)

So... to set it all up properly, I had to figure out a few things. First it took a while to get it rendering in a dark theme, I ended up installing Flatseal to change an environment variable that it seems to have set at first. Eventually I found where the config folder ended up (more my fault, I forgot there was a menu item to open it). But I am also still trying to figure out why it can't network properly, I assume some flatpak permissions thing... it doesn't seem able to access zeroconf .local addresses like my klipper host uses.

In other words: _Please_ don't get rid of the appimage... if it comes down to "we can't offer prusa connect in the appimage version" then so be it.

TheAssassin, 2024-10-03:
(Disclaimer: AppImage project maintainer and long-time PrusaSlicer user here.)

The issue appears to be related to three facts:

- a) Prusa decides not to bundle WebKit
- b) WebKit >= 4.1 apparently is not backwards compatible to WebKit <= 4.0
- c) Prusa added this "Phyiscal Printer" feature

Let's have a look at these one by one.

We probably cannot do a lot about b). I'm sure Prusa's engineering tried to work around those issues they are seeing, but have not found a solution. I may have a look at the issue tracker to understand what the problem is (maybe we can solve those problems).

Regarding c), I'm sure many people could live without this tab in PrusaSlicer 2.8+ for the AppImage only. From a product perspective, I don't think it's desirable for Prusa as a company to offer such a special build that lacks features (because that then needs to be explained to users again), but on the other hand, that can be solved by showing a short explanatory text within PrusaSlicer on said tab in the UI. People already pursue this, see, e.g., [this PR](https://github.com/prusa3d/PrusaSlicer/issues/12968).

Reversing the decision behind a) would be the way to pursue anyway, though, at least in my opinion. AppImages are designed not to depend on non-system libraries. WebKit certainly does not qualify as a system library, it's almost always installed as a dependency of some application. The AppImage really should ship it either way. It should(!) be possible given that any Electron-based AppImage and a lot of other applications do so, but I'm not familiar with PrusaSlicer's build to promise anything.

There have been attempts at improving [linuxdeploy-plugin-gtk](https://github.com/linuxdeploy/linuxdeploy-plugin-gtk) (linuxdeploy being one of many tools to make creating AppImages and other portable bundles easier) and other linuxdeploy plugins to ship WebKit. See, e.g., https://github.com/linuxdeploy/linuxdeploy-plugin-gtk/pull/37, where it is shown that the people behind [Tauri](https://github.com/tauri-apps/tauri) managed to implement a portable method to ship WebKit with the help of linuxdeploy. We know that shipping Qt's WebKit integration also works fine.

TL;DR: Prusa should try to include WebKit. That'll solve the problem and it'll make PrusaSlicer more portable. If that is too much effort, I wouldn't mind AppImages that just do not ship the WebKit-based features.

Please don't hesitate to reach out if you have questions. The AppImage folks have seen the wildest problems and in most cases, we found a solution.

vega-d, 2024-10-07:
I don't see desktop applications on linux going any other way than flatpak. It's a large ecosystem of apps that allows the developer to actually do the development and not have to do silly things with packaging for multiple distros. While having a portable application format sounds nice in theory, in reality when people install prusa slicer they usually install it for a long time since they now own a printer and use the slicer for it, so is the portable executable really necessary? 

Having it in a repository like flathub would simplify updates - instead of being nagged for updates on launch (and dismissing the update because you just want to get to slicing) it can be updated along with everything else in the software store application. 

Using flatpak's runtimes (like the gnome runtime that already includes WebKitGTK) would reduce the size of the slicer for linux users as flatpak would handle the dependencies in the runtime (you the developer develop against a certain version and the flatpak client keeps that version for the application, if multiple apps use the same runtime you only need to have the runtime once). 

I didn't even know prusa slicer does an appimage until I visited the github repository, because instead of hunting the web for executables I just looked it up in the software store which gave me the unofficial flatpak. And i've been using it for the longest time with it working just fine. While I can't say my experience is universal, it does appear to involve less friction than keeping up with updates by downloading appimages all the time. 

appimages are not really the universal "applications that run anywhere" format they claim to be if you have to package and test multiple packages for different distros. Flatpaks are, anywhere there's a working flatpak client you can run any flatpak.

novaplusplus, 2024-10-08:
> if you have to package and test multiple packages for different distros.

I mean, if you have to do that, you are building the AppImage wrong... 🤷‍♂️

johnolinda, 2024-10-22:
Hello! I am sure I am in an extremely small minority, but I am a Linux user who also uses post-processing scripts. With the Flatpak version of PrusaSlicer I am unable to run them since PS is not able to see the entire filesystem or execute arbitrary code. The Appimage version works perfectly. I understand there are packaging issues with Webkit that are probably going to lead to the Appimage being deprecated. Will there be instructions on how to properly configure the Flatpak version to run post-processing scripts?

[JimB over at the Prusa forums](https://forum.prusa3d.com/forum/prusaslicer/linux-users-not-being-able-to-use-post-processing-scripts-with-transition-to-flatpak/#post-724949) has been helping me troubleshoot and as far as we can tell, the only way to run post-processing scripts based on Perl is to basically do an install of Perl inside the Flatpak sandbox. This doesn't seem to affect Bash or Python scripts in the same way though. I would appreciate some feedback on this, and how to work around this with the Flatpak version (without doing a whole separate Perl install) if this is the way things are going moving forward.

SachCZ, 2024-10-25:
Hello @johnolinda, you should be able to achieve what you want by running slicer like this:
```
flatpak run --talk-name=org.freedesktop.Flatpak com.prusa3d.PrusaSlicer
```
and in the Post processing scripts window:
```
flatpak-spawn --host dostuff
```
where dostuff is what you had there previously.

johnolinda, 2024-10-26:
OK, just for documentation's sake, here's what I did (using Flatseal) to make this work normally through the GUI. First, I enabled the Socket > D-Bus session bus setting, then added `/tmp` to Filesystem > Other files. Then I reconfigured the script command with `flatpak-spawn --host` as recommended and everything appears to work normally now. I'll test the Flatpak and report back if I notice further issues.

eduncan911, 2024-11-21:
In light of 2.9.0-alpha's announcement that appimages are no more, this issue serves no purpose.  Closing.

closed 2024-11-21 by eduncan911

Samueru-sama, 2024-11-22:
> Using flatpak's runtimes (like the gnome runtime that already includes WebKitGTK) would reduce the size of the slicer for linux users as flatpak would handle the dependencies in the runtime

[Totally not true.](https://imgur.com/a/6JP6Wap)

CRCinAU, 2024-11-22:
And for what its worth, Cura still has an AppImage:
https://ultimaker.com/software/ultimaker-cura/#downloads

probonopd, 2024-11-26:
> And for what its worth, Cura still has an AppImage

So do the PrusaSlicer downstreams:

* https://github.com/bambulab/BambuStudio/releases
* https://github.com/supermerill/SuperSlicer/releases
* https://github.com/SoftFever/OrcaSlicer/releases

novaplusplus, 2024-12-03:
Thanks for keeping this alive and showing that it's perfectly achieveable. Haven't used PS in a while but just grabbed your build to test while I try Nobara linux (light fork of Fedora 40) + KDE, seems to be working fine. Only just looked through basic functions though, did not go through and slice anything. Infamous web interface seems to work fine, as does the new printables integration ~~(why tho)~~

46cv8, 2025-01-29:
> OK, just for documentation's sake, here's what I did (using Flatseal) to make this work normally through the GUI. First, I enabled the Socket > D-Bus session bus setting, then added `/tmp` to [...]

Hey @johnolinda I'm really new to flatpak thanks for taking the time to explain this. Can you elaborate on how to enable Socket > D-Bus . I don't see it in flatseal under prusa but I'm sure I'm just missing something.

johnolinda, 2025-02-03:
So this is what the setting looks like on my computer:

[image: image.png]

Do you see this option on your device?

46cv8, 2025-02-03:
Hey @johnolinda I managed to figure it out right after I posted the comment, so I just removed my comment sorry. Thanks so much for taking the time to come and post an image! 

> So this is what the setting looks like on my computer:
> 
> [image: image.png]
> 
> Do you see this option on your device?
> […](#)

johnolinda, 2025-02-03:
No problem, I hope it continues to work well for you!

46cv8, 2025-03-06:
> No problem, I hope it continues to work well for you!
> […](#)

Hey @johnolinda , I'm a little curious is this still working for you. After what I can only assume were some ubuntu updates I can no longer access flatpak-spawn from within the context of the prusaslicer flatpak. ```Portal call failed: Failed to start command: Failed to execute child process "flatpak-spawn" (No such file or directory)``` I'm curious if perhaps they changed some security defaults that need overriding. Your not hitting this issue are you?

johnolinda, 2025-03-06:
I’ll have to check, I just recently changed computers and haven’t finished setting the new one up. I’ll pull out the old one and update it to see. I’ll update you once I’ve checked.

46cv8, 2025-03-06:
> I’ll have to check, I just recently changed computers and haven’t finished setting the new one up. I’ll pull out the old one and update it to see. I’ll update you once I’ve checked. On Mar 5, 2025, at 7:38 PM, 46cv8 ***@***.***> wrote:
> 
> No problem, I hope it continues to work well for you!
> …
> 
> 
> 
> 
> 46cv8 left a comment ([prusa3d/PrusaSlicer#13376](https://github.com/prusa3d/PrusaSlicer/issues/13376))
> 
> No problem, I hope it continues to work well for you!
> …
> 
> Hey [@johnolinda](https://github.com/johnolinda) , I'm a little curious is this still working for you. After what I can only assume were some ubuntu updates I can no longer access flatpak-spawn from within the context of the prusaslicer flatpak. Portal call failed: Failed to start command: Failed to execute child process "flatpak-spawn" (No such file or directory) I'm curious if perhaps they changed some security defaults that need overriding. Your not hitting this issue are you?

No worries thanks! Just as a bit of background the command "flatpak-spawn" wasn't showing as valid on the host machine, so I searched and it seems it's included in the flatpak-xdg-utils package so I installed that. The install path is /usr/libexec/flatpak-xdg-utils/flatpak-spawn so it doesn't get found when calling flatpak-spawn. I tried updating the run command in Prusaslicer to point to the full flatpak-spawn path but it didn't seem to help. "/usr/libexec/flatpak-xdg-utils/flatpak-spawn --host flatpak-spawn --host /home/user/Projects/prusaslicer_scripts/make_fcp_x3g;" Not sure perhaps I need to share the folder "/usr/libexec/flatpak-xdg-utils/" with the prusaslicer flatpak as well. Anyways I'm curious if it stopped working for you too, it's not urgent though, for now I'm using using octoprint with the gpx plugin instead :)