prusa3d/PrusaSlicer #13382: pas de support et Ht variable impossible

18pool49, 2024-09-16:
### Description of the bug

une pièce sans support >> quand on demande hauteur de couche variable  le slicer prétend que des support son activé >>>
ce qui est faut  et impossible de trancher 
[image]
[image]

si je retire la hauteur variable le tranchage est possible 

### Project file & How to reproduce

[babypanda 5c ht variable impossible.zip](https://github.com/user-attachments/files/17009326/babypanda.5c.ht.variable.impossible.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1 rc1

### Operating system

w11

### Printer model

XL5

18pool49, 2024-09-16:
en version 2.8.0 la hauteur variable ne peut pas etre utilisé  car soit disant des supports serait actif 
[image]

TerPuc, 2024-09-17:
Hello @18pool49, thanks for your post. We already have this reported here: #9462 Therefore I close this issue as duplicate.

closed 2024-09-17 by TerPuc

18pool49, 2024-09-17:
non vous n'avez pas compris 
je ne fais pas usage de support >> pas du tout <<
mais avec la hauteur variable le slicer prêtant qu'il y a des supports >>> CE QUI EST FAUT <<<<<