prusa3d/PrusaSlicer #13383: Physical printer TAB on Prusaslicer 2.8.0 on MK3S and shows only Connection Failed

lezetuser, 2024-09-16:
### Description of the bug

Physical Printer tab seems to be not able to setup the connection to the local Prusalink on MK3S. 
Connect connection works fine
Physical setting connection test works fine
I am able to send a file towards a printer
From Physical printer settings I can open the Prusalink local URL site.
I have tried via API key and HTTP and HTTP Digest as well

[image]

### Project file & How to reproduce

1. Open Prusa Slicer
2. Select the MK3s Physical printer
3. Click on Physical Printer TAB
4. For a moment seems the Slicer tries to setup the connection
5. Then the below message shows up
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

W 11

### Printer model

not relevant

jamckelvey, 2024-09-22:
+1
Slicer v2.8.1
Windows 10

Robagon3D, 2024-09-25:
+1
Slicer v2.8.1
Windows 11
MK3S with RP0v2

NeonSoap, 2024-10-04:
I have the same issue. Can send files through Slicer, can access PrusaLink in a browser, but "Connection failed" in PrusaSlicer Physical Printer tab.

Slicer v2.8.1
Windows 10
MK3S+ with RPi Zero W

ncx-jag, 2024-10-21:
I too have the same issue.  I hibernate every night to take my laptop home from work.  When I return in the AM connection appears lost.  F5 to refresh does nothing.  Still can send jobs and phone app works fine.

Slicer v2.8.1
Windows 10
MK3S+ with RPi Zero W

TheBraveKing, 2024-10-25:
+1 
win 10
MINI+

petrisi, 2024-12-01:
Same behavior here, PrusaLink based MK3 printers don't work. XL and Klipper based printers work fine.

TheManFormerlyKnownAsSeth, 2024-12-03:
Same behavior, possibly different cause, migrated octoprint managing neptune 3 pro to a new install using different port. Was going to adjust here but found tab unusable.

dtoubelis, 2026-03-12:
+1
Slicer v2.9.4
Prusalink v0.8.1
Win11, Mac

PrusaLink log file shows this:

```
Mar 11 22:21:01 prusa-mk3s-01 prusa.link.web.lib.auth[665]: INFO: Digest: nonce value not match {check_digest():47}
```
...there are also some exceptions earlier in the log (not sure if they are related since everything else works fine):

```
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Was[DEBUG]: Matched RegexPairing for ^(ok.*)|(Done saving file\.)$ with priority inf having 2 handlers calling {(1921630728, 1913887984): <bound method SerialQueue._confirmation_handler of <prusa.link.serial.serial_queue.MonitoredSerialQueue object at 0x72139cf0>>, 1887832696: <function ThreadedSerialParser.decoupled.<locals>.inner at 0x70860a78>} {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Was[WARNING]: Instruction M862.2 Q did not capture its expected output, so it REFUSES to be confirmed! {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Repeat - triggered by instruction refusing confirmation. {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Listing all threads with stack traces for debugging {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Thread MainThread stack trace: {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/home/pi/.local/bin/prusalink", line 8, in <module> {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     sys.exit(main()) {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/home/pi/.local/lib/python3.11/site-packages/prusa/link/__main__.py", line 259, in main {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     retval = daemon.run() {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/home/pi/.local/lib/python3.11/site-packages/prusa/link/daemon.py", line 58, in run {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     self.prusa_link.stopped_event.wait() {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/threading.py", line 622, in wait {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     signaled = self._cond.wait(timeout) {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/threading.py", line 320, in wait {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     waiter.acquire() {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:  {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Thread httpd stack trace: {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/threading.py", line 995, in _bootstrap {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     self._bootstrap_inner() {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     self.run() {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/threading.py", line 975, in run {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     self._target(*self._args, **self._kwargs) {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/home/pi/.local/lib/python3.11/site-packages/prusa/link/web/__init__.py", line 92, in run {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     self.httpd.serve_forever() {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/socketserver.py", line 233, in serve_forever {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     ready = selector.select(poll_interval) {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/selectors.py", line 415, in select {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     fd_event_list = self._selector.poll(timeout) {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:  {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Thread Thread-1 (_run_loop) stack trace: {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/threading.py", line 995, in _bootstrap {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     self._bootstrap_inner() {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:     self.run() {_confirmed():450}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING:   File "/usr/lib/python3.11/threading.py", line 975, in run {_confirmed():450}
[... cut ...]
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Was[DEBUG]: Recv: 302 {_read_continually():327}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Was[DEBUG]: Matched RegexPairing for ^(?P<code>\d{3,5})$ with priority 1773278924.1355045 having 1 handler calling {(1955861000, 1877902832): <bound method MatchableInstruction.output_captured of Instruction 'M862.2 Q'>} {fire():52}
Mar 11 21:28:44 prusa-mk3s-01 interesting_logger[665]: WARNING: Was[DEBUG]: Recv: ok {_read_continually():327}

```

closed 2026-09-29 by prusaslicer-bot[bot]