prusa3d/PrusaSlicer #13384: No Sparse Layers still travels each layer - 2.8.1-rc1

halofx, 2024-09-17:
### Description of the bug

When selecting No Sparse Layers for the wipe tower on a multi color print, the toolhead still travels to the location of the purge tower on each layer.

[image]

### Project file & How to reproduce

add object to print that only has a few color changes, or manually set extruder changes in slice preview.
Use a stock Prusa MK3.9 speed profile,
Select No Sparse Layers In Multiple Extruder Print Settings.
Slice, and preview with travel moves showing
[Sparse_layer_travel_MK4.zip](https://github.com/user-attachments/files/17023257/Sparse_layer_travel_MK4.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1-rc1+win64

### Operating system

Win10 Pro 22H2 19045.4651

### Printer model

Prusa MK3.9

TerPuc, 2024-09-17:
Hello @halofx , thank you for reporting this. We will look at it.

Issue created: SPE 2483

lukasmatena, 2024-09-19:
Fixed in [2.8.1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1). Closing.

closed 2024-09-19 by lukasmatena