prusa3d/PrusaSlicer #13386: Wrong filament selected in Gcode for MMU3

VH-fixer, 2024-09-17:
### Description of the bug

When I select Filament 2, or T2,  T1 is output in the G Code.
If I initialy had selected filament 1, selecting filament 2 does not cause the Gcode to update,   The Gcode was regenerated with the "SLICE" selection. 

### Project file & How to reproduce

Example Gcode and .3mf file included.   The second filament is selected but T1 is output in the Gcode. 
[2.8.2-wrong-filament-mmu3.zip](https://github.com/user-attachments/files/17029264/2.8.2-wrong-filament-mmu3.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1alpa1

### Operating system

Windows 11 23H2

### Printer model

MK4 with MMU3

neophyl, 2024-09-17:
If I recall correctly aren't tools numbers from zero ?  Same with filaments etc.  So T0, T1, T2.  It is quite common in code for an index to start from 0 and for the 'user' side display to start from 1.

VH-fixer, 2024-09-17:
T0-T4 correct.  Sorry I have been conditioned to Gcode that followed standards over the last 45 years.  T0 is  unload, spindle on NC machines.  Also filament 1-5 is used in the slicer, not 0-4.  Maybe it should be 0-4 to match the digital T code?
There is a problem.  

I selected Filament 2 in the slicer and the MMU used Filament 1, T0.    In trying ot re-open the .3mf and reproduce it correct the T address. 

The uploaded .3MF file is what was used and filament selection 2 when I printed the part.  At the printer, T0 was used.  I pulled this off of the USB drive of the printer that was sent directly from the PrusaSlicer 2.8.1Alpha1

[image]

[xlcd-bumper back_0.4n_0.2mm_PETG_MK4ISMMU3_1h5m.zip](https://github.com/user-attachments/files/17029723/xlcd-bumper.back_0.4n_0.2mm_PETG_MK4ISMMU3_1h5m.zip)

Getting this to recreate and then be able to submit it so it recreates for others is a problem.  Especially when it looks like re-opening may clear it.  I have double checked everything and it happens.   It needs to work 100%.  I shouldn't have to check the tcodes all the time in the program.    This is not the first time this has happened.   It has come and gone in previous versions of the slicer.

closed 2026-09-29 by prusaslicer-bot[bot]