prusa3d/PrusaSlicer #13395: Please revert the Menu menu change

jdsheneman, 2024-09-19:
Prusa Slicer 2.8 moved the menu to a separate Menu menu.  This requires an extra click or keystroke for every operation.  Please revert this change.

JohnOCFII, 2024-09-19:
Check out the release candidate for 2.8.1. Prusa reverted that change.

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1

Snuff1eupagus, 2024-09-19:
I would comment that even though the dev team changed the button, the Settings selection is taking up an unreasonable amount of real-estate for that one selection and needs to be moved inline, with other menu items. There are still several +  major problems with 2.8.1 rc1 however it is such a massive improvement to the crap that is 2.8.0 final that I think some users may have lost sight.

TerPuc, 2024-09-19:
Hi @jdsheneman, this is a duplicate of closed issue #12943 therefore I close it.

closed 2024-09-19 by TerPuc