prusa3d/PrusaSlicer #13398: Abrasive Material and High Flow Nozzle Issues with Prusa Connect

chris-5634, 2024-09-19:
### Description of the bug

When trying to export gcode through prusaconnect, warning that filament is abrasive and highflow nozzle is recommended appear, even though settings are checked off.  

Does not impact uploading file, but can be confusing. 

I have access to MK4 with MMU and without, along with a 5T xl, issue not present on regular MK4 when sliced as single extruder, but remains on the MMU printers and the XL.

### Project file & How to reproduce

[Connect Issues.zip](https://github.com/user-attachments/files/17065578/Connect.Issues.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.1+win64-202409181359

### Operating system

Windows 11 Pro Build 22H2

### Printer model

Prusa MK4 with MMU3

ChainVTF, 2024-09-19:
I am having the same issue with 2.8.1, I also have a MK4 w/MMU and a 5T XL.

kvtorp, 2024-09-20:
Same issue, with 2.8.1 and XL 5TH, MK35
[image]

spkane, 2024-09-20:
Similar issues here:

[image]

edooper, 2024-09-21:
Same issue here:

MK3.5 + MMU3
FW 6.1.3+7898
Prusaslicer 2.8.1 MacOS-arm64

Standard filament profiles

petrisi, 2024-09-21:
5T XL here, same issue. Started after 2.8.1 update (never seen these warnings before.)

davbow123, 2024-09-25:
Ditto 
[image]

KlausKraemer, 2024-09-25:
[image]

I cannot repeat the error at the moment, bit on former occasions Slicer thought I had a 1mm nozzle in my XL - but all nozzles are 0.4 mm, I checked.

KlausKraemer, 2024-09-26:
Got the false warning again. The information chain "Slicer - Prusa Connect - XL5T" seems to be broken and there is no information about nozzle sizes available:

[image]

Translation: _Werkzeug_ = Tool - _Düsendurchmesser_ = nozzle diameter

Beside that: printing is absolutely fine until now.

closed 2026-09-29 by prusaslicer-bot[bot]