prusa3d/PrusaSlicer #13402: Aligning tools for different objects

3dGeneDesigns, 2024-09-20:
**Describe the solution you'd like**
I think there should be a few alignment tools for different objects. Currently as it sits, it's mostly just measuring distances and adjusting till it's lined up properly. 

**Describe how it would work**
The way I imagine it would be how assembly mates work in most 3d Cad programs. These let you align via faces, offsets/distances, concentric, or coinciding edges/faces. The measuring tool already is great at detecting flat surfaces and circles as holes, so it'd be one step further on to that I'd think. Maybe even giving each object a centered 3 planes to give further alignment options.

This would make customizing things with different objects easier. I can see this being extended to text and SVGs as well.

Had this idea when I was creating a customizable slider puzzle with a negative volume object added. 

Sorry if this has been suggested before.

neophyl, 2024-09-20:
Lol, you might say its been suggested before.

https://github.com/prusa3d/PrusaSlicer/issues/13073
https://github.com/prusa3d/PrusaSlicer/issues/13035
https://github.com/prusa3d/PrusaSlicer/issues/12891
https://github.com/prusa3d/PrusaSlicer/issues/11220
https://github.com/prusa3d/PrusaSlicer/issues/9453
https://github.com/prusa3d/PrusaSlicer/issues/7132
https://github.com/prusa3d/PrusaSlicer/issues/4501
https://github.com/prusa3d/PrusaSlicer/issues/1425

root-hal9000, 2024-11-27:
> Lol, you might say its been suggested before.
> 
> #13073 #13035 #12891 #11220 #9453 #7132 #4501 #1425

I am surprised this doesn't get more love from the team - I could be completely wrong, but it seems like most of the underlying pieces for it are already there, and it would be soooo helpful, especially for aligning /centering modifiers

WhiteDragonD, 2025-07-07:
Bumping this up because I had to use Fusion and spent a long time just trying to import/align parts..... So this would be SOOOO HELPFUL if it can be done in the program and takes only a few clicks.

Jan-Soustruznik, 2025-08-07:
closed as duplicate of #1425

closed 2025-08-07 by Jan-Soustruznik