prusa3d/PrusaSlicer #13406: PrusaSlicer 2.8.1 crashes upon closing Win 7 64

kcgit007, 2024-09-20:
### Description of the bug

Prusa Slicer 2.8.1 crashes upon closing on Win7 64.
**Crash report:**
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	prusa-slicer.exe
  Application Version:	2.8.1.0
  Application Timestamp:	66eaddda
  Fault Module Name:	PrusaSlicer.dll
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	66eaddd2
  Exception Code:	c0000005
  Exception Offset:	0000000002aa0cfc
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1031
  Additional Information 1:	9745
  Additional Information 2:	974561830ec6d90f907490c66312ff74
  Additional Information 3:	4b75
  Additional Information 4:	4b7541fc38842f3043c81282874197f6

### Project file & How to reproduce

To reproduce: 
start Prusa slicer
go to menu File -> Exit 
See "PrusaSlicer has stopped working" dialog.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.1+win64

### Operating system

windows 7 ver. 6.1.7601 SP1 64

### Printer model

MK3S+

Snuff1eupagus, 2024-09-21:
Every time for me too, if you open with the prusa-slicer-console and close by shutting down the prusa-slicer-console it wont crash, and you can see all the errors it loads with when starting. It's been this way since the alpha release. Someone probably reported this months ago.

friendlyFriend4000, 2024-09-22:
I thought Windows 7 is not supported anymore

Snuff1eupagus, 2024-09-25:
That's a Loser observation. I'm sad of this and anyone that subscribes to it. do the numbers
I use 7 64 bit, 8.1 10 and 11 all are impotent and relevant to this date. Remove one currently and lose printer sales.

Think it through. want to argue the subject?

Hanier, 2024-10-25:
PROBLEM:
APP Crash after hitting the "Close Program X" Button.

DESCRIPTION of BUG:
From Windows Fault applet

Problem Event Name: APPCRASH
Application Name: prusa-slicer.exe
Application Version: 2.8.1.0
Application Timestamp: 66eaddda
Fault Module Name: PrusaSlicer.dll
Fault Module Version: 0.0.0.0
Fault Module Timestamp: 66eaddd2
Exception Code: c0000005
Exception Offset: 0000000002aa0cfc
OS Version: 6.1.7601.2.1.0.256.48
Locale ID: 1033
Additional Information 1: 9745
Additional Information 2: 974561830ec6d90f907490c66312ff74
Additional Information 3: f698
Additional Information 4: f6983117dd7a0dca09b50ec404d6451c

TO REPRODUCE:
Attempt to close PrusaSlicer in the normal way. Hit the Red-X in the Upper Right of the window.
Windows reports the fault asking if I want to check online and close, or just close.

APPVER:
PrusaSlicer 2.8.1

OSVER:
Win7 Ver6.1 (Build 7601) ServPack 1

PRINTER: its off by the way
MK4 + Kit build (Working Great, LOVE-IT)

closed 2026-09-29 by prusaslicer-bot[bot]