prusa3d/PrusaSlicer #13408: The M486 not present in Marlin2 gcode flavor

how-eee, 2024-09-21:
### Description of the bug

The M486 (cancel objects) support is implemented but not added to Marlin2 gcode flavor.
It is available in RepRap g-code flavor.

Since many printers use marlin 2+, Please add this feature vhen Marlin2 g-code flavor menu selected.

[image]

### Project file & How to reproduce

1.choose marlin2 gcode flavor
2. slice any file with 2 or more object
3. check gcode for NO m486 presence
4. choose reprapfirmware gcode flavor
5. slice any file with 2 or more object
6. check gcode for YES m486 presence

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

win 10  64

### Printer model

skr 1.3 marlin2+

how-eee, 2024-09-21:
Im not sure if I lost some features of Marlin2 if I use Reprap gcode flavor

rtyr, 2024-09-30:
Marlin2 gcode flavor is supported and it generates M486. Are you sure you have it enabled in print settings? Please provide your project file.

closed 2024-11-04 by rtyr