prusa3d/PrusaSlicer #13410: Seam does not remain within painted seam boundary when selecting random seam.

jurassic73, 2024-09-21:
### Description of the bug

While having Advanced: Seam Position: Random selected, I expected my seam to remain within the seam painting width/highlighted area but it goes outside that boundary. Also, a modifier to select seam type by height would be awesome.

[image]

[image]

__

### Project file & How to reproduce

[test_pillar.zip](https://github.com/user-attachments/files/17084692/test_pillar.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Win10

### Printer model

MK4s

Snuff1eupagus, 2024-09-24:
I'm not a good teacher, but I can assure you all you have suggested and asked for is possible.
[image]
9 out of the 12 cylinders shown are variations on your originally posted file, using the brush tools in slic3er.
I think you need to spend more time learning to use the brush tools. the last three where me embellishing and showing even  more control then you asked for. Displaying multiple types of seam on a single model is also currently possible and has been for years. I will admit that PrusaSlicer 2.5 - 2.8.0 has made this incredibly difficult. But it dose become easier since 2.8.1 hopefully it will stay that way the developers have a way of screwing things up and not caring. Normally this goes unnoticed since the majority of users don't know what slicer can do.  Good luck learn hard, and this is already possible, and has been for years. At least since 2.3

I'm sure a hundred People here can show you how to do this.

ENJOY.

Jan-Soustruznik, 2024-10-04:
Hello, @jurassic73 , thank you for your feedback. I created internal ticket: SPE-2510. We will try to fix in some next releases. Have a nice day

closed 2025-02-18 by lukasmatena