prusa3d/PrusaSlicer #13411: Ability to change orientation of infill pattern

truegret, 2024-09-21:
**Describe the solution you'd like**
As 3D model creator and print user that struggles with optimizing print orientation to maximize part strength, I want the ability to specify the direction/orientation of the infill pattern.

**Describe how it would work**
I imagine a dialog box with a new parameter in the Print Settings | Infill | Advanced section named Infill Axis. This would be a drop-down box with a choice of the 3 (three) axis X, Y, Z.  I imagine the Print Settings | Infill | Advanced | Fill Angle numeric input box be the infill angle that generates the infill "around" the selected Infill Axis. For example, when I choose Infill Axis = "Z", and specify an angle in the Fill Angle it would behave as it does currently.  If I choose Infill Axis = "Y", the Infill Pattern would be generated along the "Y-axis" according to the Fill Angle around the "Y-axis".

**Describe alternatives you've considered**
The alternatives I currently have available is to orient the Model on the build plate differently. This is less than ideal since the features of the model may not allow for a stable print.

**Additional context**
The need or reason for this request can be visualized with this example. Imagine an elongated model shaped like a "C", where the body thickness is 25mm X 25mm and is 100mm long. It is not desirable to print the model oriented as a "C" on the bed because this would require supports, and it is ideal to have this model printed on its side for strength (no layers along the long side). However, without the feature requested, the infill pattern is along the X-Y axis in the part, and would be desirable along the length.

[image]

Lukrative525, 2024-10-01:
Not sure if this is what you're trying to do, but PrusaSlicer already allows users to set the fill angle:
[image]
[image]
[image]
[image]

truegret, 2024-10-01:
Thank you, @Lukrative525, for your help and effort. I have been struggling with how to best communicate this, and I seem to be failing with words. So, I created a "quick" F360 split-body representation of what I mean. I want to lay the part on its side as you indicate, however, I wish to be able to "rotate the infill 'pattern'" along the X or Y axis. Such that the grid is not developing along the Z-axis but along the X or Y. 

Does this make sense?:

[image]

Thanks again all, and my apologies if this discussion is in the wrong thread. Just tell me if it should be somewhere else.

Lukrative525, 2024-10-01:
Ah, that makes sense. I haven't looked under the hood, so to speak, so I don't know how PrusaSlicer handles this, but my guess is that adding the ability to essentially rotate the infill pattern about an arbitrary axis (rather that just z) would require non-trivial refactoring, but I could be wrong.

You want this functionality in order to optimize your parts for strength? Unfortunately you're probably not gonna get a whole lot of strength by doing as you propose. For the effort required, you'd be better off just rotating the infill about the z-axis so that the traces are aligned with the direction of the anticipated stress. Even better, just add a few perimeters. Even better-er, model in sub-sections of your model where you want more strength, set the subsection as a modifier and then increase its density to 100. I actually did this recently with an engineering model (see attached pictures).

Alternatively, you could propose a new infill pattern that is built like you want. Then you could just rotate about z as usual to get it facing your preferred direction.

It's certainly an interesting feature request!

[image]
[image]
[image]

truegret, 2024-10-02:
Yea, I hear ya'. I have not taken a dive into the code to see how infill is handled. I suspect you are correct about a "no small feat" to refactor the code. It seemed like a useful feature - potentially. WRT strength, I sometimes use a relatively affordable Carbon Fiber PLA from ZYLtech ($26/kilo US) because it is cool, feels good, is light-weight. My thinking was to have the fibers run the length to add strength (ideally).

This part, at the start of the request, is a piece that attaches to the eaves of a house (top of the pic) and the other end (with the four hex holes) is for a PZT camera. So the bottom cantilevers from the eaves and holds the camera. I did this for a silly reason. The camera pans 300º. With the standard mount, I lose 60º of view, and have to slew around viewing the house to get to the other side (if the camera is at 0º I have to slew around, viewing the house, to get to "the front" at 300º).

I believe you are correct in that I could stiffen the corners.  I think I'll go in to F360 and generate my own "Infill" along the Y-axis ..... 

Thanks for your help and the tips! Cheers

closed 2026-09-29 by prusaslicer-bot[bot]