prusa3d/PrusaSlicer #13413: 2.8.1 Crashes frequently when editing text

mkoistinen, 2024-09-22:
### Description of the bug

**What is the problem**: PrusaSlicer crashes frequently when editing text objects. It seems to more regularly happen when clicking "Use surface", but not always, also seems to happen a lot when editing the depth widget. However, crashes are not exclusive to editing these two settings. I also noticed that the orientation of the local origin of the text modifiers often change from being parallel to the face they're on to something more parallel to the bed.

**What did I expect**: PrusaSlicer to be at least as stable as 2.8.0.

**Screenshots**: They aren't really possible - the app just disappears (crashes) arbitrarily while editing text. But, perhaps this is helpful:

[image]

Note that the first crash happened while I was adding the second text modifier, then had regular crashes after that. I gave up using 2.8.1 after the 10th or 11th crash. I ended up using 2.8.0 to finish my edits (was perfectly stable there!), then, had to bring it into 2.8.1 again, because 2.8 couldn't slice due to issues with Abrasive Nozzle support introduced in 2.8.1.

### Project file & How to reproduce

Not sure how helpful a file will be due to moving back and forth from 2.8.1 and 2.8.0. Please not this happened in 2.8.1 before any completion of my work in 2.8.0. The typeface I use is a TrueType font called "Gotham Black", but, I doubt I can include it here because it is a copyrighted font available under a purchased license only.

[14-sided-rounded-die.3mf.zip](https://github.com/user-attachments/files/17087536/14-sided-rounded-die.3mf.zip)

Load a 3D model. Attempt to create a text modifier. Add some text and set the height to 10.7 and the depth to 1.5. Also check the box for "Use Surface" and choose the "Modifier" radio button. Rotate to suit.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.1+MacOS-arm64

### Operating system

macOS 15 Sequoia

### Printer model

Prusa XL (5 head)

makama80, 2024-09-30:
I can confirm the issue. I have noticed it the first time when using my MK4 with  PrusaSlicer 2.8.1 under MacOS Sequoia 15.0.
Of course I have reinstalled Prusa Slicer (after a complete removal using AppZapper) but the issue remains.

I have attached the MacOS crash report.

[Prusa 2.8.1 MacOS Crash report.txt](https://github.com/user-attachments/files/17184238/Prusa.2.8.1.MacOS.Crash.report.txt)

bene-K, 2024-10-01:
Same here. As soon as I add Text to an Object (set on cut) and enable "on surface" PrusaSicer crashes. I used the feature before without an issue.
Sorry, but can't provide the specific file.

PrusaSlicer 2.8.1+win64
Build:     PrusaSlicer-2.8.1+win64-202409181359
Windows 10 (build 19045), 64-bit

feesta, 2024-12-03:
Also having a crash when editing text as an added 'part' to an object. I hit this on both 2.9 and 2.8.1.

[Prusa 2.9 alpha crash details.txt](https://github.com/user-attachments/files/17999412/Prusa.2.9.alpha.crash.details.txt)

Bat-Maker, 2025-02-20:
Still an Issue on 2.9 using M1 Max MacBook Pro & M4 Pro Mac Mini. Trying the Alpha Release as a Hail Mary here.

Bat-Maker, 2025-02-20:
Just confirmed, still occurs on 2.9.1 Alpha.

[Crash report.txt](https://github.com/user-attachments/files/18880566/Crash.report.txt)

mrpajzl, 2025-03-04:
Still happens and it is pretty annoying. Cannot share the files since it is a premium model but maybe in a private message it would be fine for the sake of debugging on the side of Prusa research since it is only one part out of 16 for that model

mtbikemarc, 2025-05-10:
This happens on Prusa and Orca. Very annoying. Could someone please fix it :-).

mrpajzl, 2026-09-02:
The same still happens on the 3.0.0 version. I add any text and configure it to copy the surface shape and the whole slicer crashes