prusa3d/PrusaSlicer #13415: Mesh export from Fusion360 not received by PrusaSlicer - MacOS ARM64

PearlBlueSoul, 2024-09-22:
### Description of the bug

Since updating to Version 2.8.1+MacOS-arm64, When exporting mesh from Fusion360 as STL, PrusaSlicer no longer receives the STL output from Fusion.  I have reselected the application in Fusion, issues persist.   

### Project file & How to reproduce

No project can be created since no models are being received. 

https://github.com/user-attachments/assets/e0ba4eb2-ec35-4d5d-be6f-17ad3a2deab1

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.1+MacOS-arm64

### Operating system

MacOS 14.7 (23H124)

### Printer model

Original Prusa i3 MK3

Jahor, 2024-09-23:
It looks to be something with Fusion update. It worked for me on PrusaSlicer 2.8.1 till Fusion updated to 2.0.20256.
Now it only works if PrusaSlicer is not running when I export model from Fusion.

blalor, 2024-09-24:
I noticed this with PrusaSlicer 2.8.0, as well.  It probably _is_ a Fusion issue, but PrusaSlicer does come to the foreground, which makes me think PS bears some responsibility, as well.

PearlBlueSoul, 2024-09-24:
That’s kind of a tough part for me because today, I did notice that it did sporadically seem to work, and then I realized after grinding out why it only worked when Prusa was not open already.  

Also a tough call when both things seem to change at the same time. I’m not exactly sure which one created the problem first, but I did not have this problem until I updated Prusa, and  since Prusa is receiving data from Fusion, one could think that’s where the problem lies.  I also did not have this problem on the prior version, although I don’t believe it was 2.8.0. It was probably older than that.  

I have not yet tried going back a few versions to see if maybe that works better. There was no inherent reason I updated other than just to get rid of the alert.

crabbedh, 2024-09-26:
Absolutely it is Fusion 360. It worked fine for me with Prusa Slicer 2.8 and then only broke when Fusion updated within the last few weeks (mid Sept, '24).. Fusion required me to setup the whole link to PSlicer again. I can verify that if Prusa Slicer is open, it will not send across STLs (solid or mesh, same failure). But if Prusa Slicer is closed, it will send it along.

praegustator, 2024-10-13:
According to the [changelog](https://www.autodesk.com/products/fusion-360/blog/september-2024-product-update-whats-new/) from October 1, 2024, it has been fixed. 

I can confirm the issue is gone with the latest Fusion 360 update (currently it is 2.0.20476).

PearlBlueSoul, 2024-10-13:
Can confirm it’s working again for me too, apologies for the false alarm! Thank you, everyone for your input!

closed 2024-10-13 by PearlBlueSoul