prusa3d/PrusaSlicer #13416: You don't need to wait for bed and extruder temperatures when only homing.

skoblins, 2024-09-23:
### Description of the bug

It's an optimization proposal, that I use on a daily basis, with my Prusa + MMU3, and I can recommend it.
I have a ready commit with the necessary changes, but I can't push my branch to the repo...
Homing lasts several seconds. Opposite to the leveling case, the temperatures are not necessary for homing.
So we can make the heatup in parallel. It saves a lot of user time, when he/she restarts the prints often.
I use it, I have tested it with MMU3, and there are no issues with this approach (?).

### Project file & How to reproduce

Every restart of the print.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0+linux-x64-GTK3-202406270929

### Operating system

linux

### Printer model

Prusa MKIV + MMU3

rtyr, 2024-09-23:
We are not waiting for bed temperature before homing. We are only waiting for specific hotend temperature, which is intended to prevent incorrect Z leveling height and to prevent damage to the sheet over time (caused by either too low or too high temperature).

skoblins, 2024-09-23:
Sorry I wasn't specific enough :)
I refer only to the Start G-code, when the hotend is usually not very hot
even if you restart prints right away.
You are right about the bed temp.

rtyr, 2024-09-24:
Yes, but as I mentioned, we don't want too low temp (cold hard filament potentionally sticking out of the nozzle), nor too high temperature (not cooled enough from whatever was printer doing before). Both scenarios are problematic (and both were tested during MK4 development). I can understand your reasons, but we will leave it for now.

closed 2024-09-24 by rtyr

skoblins, 2024-09-24:
That would mean that it could still heat up during the XY travel, and then
only wait a few seconds before the last Z-move.
That intelligently spares more than 10s (? would need to count....) on each
start.