prusa3d/PrusaSlicer #13418: Crash when generating wipe tower with non generated enforced supports

murk-sy, 2024-09-23:
### Description of the bug

When slicing a file with thin overhangs that do not generate enforced supports, having multimaterial raft (or perhaps supports below those overhangs?) causes a consistent crash when slicing with wipe tower enabled.

There are some peculiarities in the crashes that seem inconsistent, but I named the bug as best I could.

A 3mf for easy reproduction is attached below.

### Project file & How to reproduce

[crash-overhang.zip](https://github.com/user-attachments/files/17095441/crash-overhang.zip)

1. Open, set "Raft layers" to 0. File slices fine. Note that no supports are generated.
2. Open, set "Supports" to "None", and remove Paint-on supports. Crash
3. Open, change Print settings to "0.20mm STRUCTURAL" and Transfer the settings. Supports are set to None. Slices fine.
4. Open, set "Support material / raft interface extruder" to 0. File slices fine, no supports are actually generated.
5. Open, select the lower shape-box, unlock scaling and set X to 4mm. File slices fine, supports are generated.
6. Open, disable "Wipe Tower". File slices fine, no supports are actually generated.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 10

### Printer model

Prusa XL

Jan-Soustruznik, 2024-10-04:
Hello, @murk-sy, Thank you for your feedback, but I cannot evoke any crash that you mention above. Could you create a 3mf that will evoke the crash? Maybe I did something wrong, but I tried a lot of steps, but any crash wasnt evoked. Thank you, have a nice day

murk-sy, 2024-10-04:
I must have accidentally saved a version with rafts disabled, whoops. Must've forgotten among all the crashes.
After each opening, do the following:
Print settings > Support Material > Raft > Raft Layers: set to 2 layers

Then the listed reproduction steps should crash reliably.

Jan-Soustruznik, 2024-10-07:
Hello, @murk-sy, now I can reproduce on our side.I created internal ticket SPE-2511. Thank you for your investigation. Have a nice day

lukasmatena, 2025-02-18:
@murk-sy Thanks for the heads up. I am glad that it works for you now, but unfortunately the 3MF we attached to this problem in our internal tracker still crashes. The problem is certainly UB of some sort, and some other changes may have masked it in your case, but I would bet that the underlying cause is still there. If you don't mind, I will let the ticket open for now.

murk-sy, 2025-02-18:
Thanks for the feedback, will leave it for you guys to close.

closed 2026-09-29 by prusaslicer-bot[bot]