prusa3d/PrusaSlicer #13419: Supports are sliced in the air

AlexStOd, 2024-09-23:
### Description of the bug

In some cases supports have gaps between layers so a next layer is placed in the air, not based on the previous layer.
Example:
[Case.zip](https://github.com/user-attachments/files/17096617/Case.zip)
Here layer 4 is OK:
[image]
But layer 5 is NOK (several elements arn't above the previously printed supports):
[image]
Layer 6 is strictly NOK (big element is printed in the air): 
[image]
How it looks from the bottom:
[image]

### Project file & How to reproduce

[Case.zip](https://github.com/user-attachments/files/17096662/Case.zip)
1. Open 3mf file from attachment
2. Slice
3. See error

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11 Pro

### Printer model

Sapphire with klipper

neophyl, 2024-09-23:
Duplicate issue.

https://github.com/prusa3d/PrusaSlicer/issues/9968
https://github.com/prusa3d/PrusaSlicer/issues/12418
https://github.com/prusa3d/PrusaSlicer/issues/9601 
etc

u89djt, 2024-09-23:
@neophyl this looks different to those three - they have an island appearing suddenly in mid-air. This issue is about a loop (continuing upwards) of the trunk of the support being drawn out over nothing, like it's been pushed aside by that protrusion in the body of the print.

neophyl, 2024-09-23:
This has 100% been reported before though.  I'll try and find the specific report I remember seeing.

Ah ah, found the one I was thinking of https://github.com/prusa3d/PrusaSlicer/issues/10725  teach me to do a too quick search during lunch.  It is still open too.  Most/all of the other ones I remember seeing have been closed in favour of that one it looks like.

Jan-Soustruznik, 2024-09-26:
Closed as duplicate https://github.com/prusa3d/PrusaSlicer/issues/10725

closed 2024-09-26 by Jan-Soustruznik