prusa3d/PrusaSlicer #13420: Snapmaker 2.0 Printing Defects - Z Feed Rate too high

jalapenopuzzle, 2024-09-23:
### Description of the bug

I have experienced several issues using PrusaSlicer 2.8.0 with Snapmaker 2.0 A350T.

1. First layer speed 50mm/s is too fast and has poor adhesion. I reduced it to 20mm/s and had much better results, but I did not tune this value.
2. First Layer Height 0.3mm seems to be too high. I had better results with 0.2mm.
3. I have had HUGE issues with missing Z steps while printing. PrusaSlicer produced the following benchy:
[image]
This benchy is only 35mm high (should be 48mm high).
I did a lot of testing with this one.
The Snapmaker 2.0 uses linear actuators with different screw pitches per axis:
<https://shop.snapmaker.com/products/linear-module-with-tmc2209-a350t-f350>
The consequence of this is that the Z axis has a lower feed rate limit than the X and Y axes.
Feed Rate Limits (reference: Snapmaker's Luban slicer v4.13.0):

- X+Y: Slow: 40mm/s Medium: 50mm/s Fast: 60mm/s
- Z: 10mm/s (F600) - (Reference: Luban generated G-Code)

PrusaSlicer generates G-Code with Z feed rate of F7200 (120mm/s). It appears that this is causing the Z axis to skip steps.
PrusaSlicer appears to have the following parameters affecting the z speed:

- Print Settings -> Speed -> Speed for non-print moves -> Travel (default: 120mm/s -> F7200)
- Print Settings -> Speed -> Speed for non-print moves -> Z travel (default: 0, causes "Travel" value to be used)

I found that setting 10mm/s for the Z travel speed had no effect. PrusaSlicer continues to generate G code with F7200 on Z movements.
The only way that I could reduce the Z speed was to reduce the Travel speed.

I performed some manual Z movement tests with increasing speeds such as:
G1 Z50 F1000
G1 Z150 F2000
I found that the speed (and stepper tone) did not seem to increase after about F3400 (56.7mm/s) (but I only tested in units of 100).

It appears that faster speeds are coming to the Snapmaker 2.0 with the vibration compensation firmware currently in beta testing <https://forum.snapmaker.com/t/vibration-compensation-firmware-open-beta-for-2-0-series-a-at-f-models/35729>. I mention this because they claim a speed increase, depending on which linear modules (20mm or 8mm lead) are installed, from 100mm/s (no VC) to 120mm/s (with VC). I don't think this applies to Z axis.

### Project file & How to reproduce

[20240921_3DBenchy_One_Colour_Z_Speed_10mm_s.zip](https://github.com/user-attachments/files/17097736/20240921_3DBenchy_One_Colour_Z_Speed_10mm_s.zip)
I sliced and exported the G-Code.
To save time I have been checking the G-Code using sed:
sed --quiet -e "s/^G1 .* Z\(-\?[.0-9]\+\).* F\(-\?[.0-9]\+\)/\2/p" file.gcode | sort -u
This prints the Z feed speeds used in the G-Code file.
I get:
F200
F7200
The F200 comes from the initial movements to approach the bed.
F7200 is used while printing, including for Z-hop movements.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Ubuntu 22.04

### Printer model

Snapmaker 2.0 A350T with upgraded linear modules TMC2209

rtyr, 2024-09-24:
3rd party profiles are usually provided/maintained by the community. The last Snapmaker bundle was provided by @macdylan. @macdylan do you have any comments regarding this issue (1st layer height/speed etc.)?

There is indeed a bug regarding some Z travel commands, we will look into that. That being said, max. Z travel speed is supposed to be primarily limited by the firmware. I would expect appropriate limits in official Snapmaker FW.

I am not familiar with Snapmaker's FW, but if it's based on Marlin, you may try to add M203 Z10 command or set machine limits to "emit to g-code".

jalapenopuzzle, 2024-09-25:
I have prototyped a solution using GNU Octave.
This program provides a reference implementation for computing the speed of a G1 move subject to the configured speed and speed_z.
It also includes a unit test.
[Test_Z_Speed.m.txt](https://github.com/user-attachments/files/17139430/Test_Z_Speed.m.txt)
Note that the file has been renamed to allow it to be uploaded to github.com. To use it with octave, remove the .txt extension and keep the .m extension.

When I run this example in GNU Octave, it produces the following output:
```octave
>> Test_Z_Speed
Computing TestAtSpeeds(120.000000,0.000000).
Completed TestAtSpeeds(120.000000,0.000000).
Computing TestAtSpeeds(120.000000,10.000000).
Completed TestAtSpeeds(120.000000,10.000000).
Computing TestAtSpeeds(120.000000,50.000000).
Completed TestAtSpeeds(120.000000,50.000000).
Computing TestAtSpeeds(50.000000,100.000000).
Completed TestAtSpeeds(50.000000,100.000000).
Computing TestAtSpeeds(100.000000,100.000000).
Completed TestAtSpeeds(100.000000,100.000000).
>>
```
I believe this code fixes the issue with the faulty computation of the `factor` in the existing code.
I have not yet attempted to correct the code in `GCodeWriter::get_travel_to_xyz_gcode()`.

jalapenopuzzle, 2024-10-05:
I modified PrusaSlicer to implement the speed limitation scheme described above, but it did not fix the problem. My Snapmaker machine still skips Z steps and produces prints that are too short.
I modified PrusaSlicer a second time to separate the Z travel moves from the XY travel moves. My Snapmaker 2.0 machine produces decent prints in this configuration.
Note that Snapmaker Luban emitz Z travel moves separately from XY travel moves. Perhaps there is a defect or an implementation limitation in the Snapmaker firmware relating to fused XYZ moves, which Snapmaker Luban works around by separating Z moves.
I will do some more experiments to determine exactly what G-code causes this issue.

jalapenopuzzle, 2024-10-09:
Commit 85332f6 allows me to print properly on my Snapmaker 2.0.
It is a PROOF OF CONCEPT change only. It is NOT suitable for merging.

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

jalapenopuzzle, 2025-01-30:
I see that this issue has been partially addressed by branch ms_fix_xyz_travel_speed (SPE-2493) merged at commit ID 6d19776942f0ea059292bd3e310764a1f9ce3d22.

I believe there are two aspects to the Skipping Z Steps problem:
1. Incorrect `from` position leads to incorrect movement vector
2. Z Speed limitation is computed without regard for vector speed decomposition according to the movement unit vector.

While looking into these issues, I found a third issue:
3. EPSILON is inconsistent with XYZF_EXPORT_DIGITS
This causes some movements to be emitted (unnecessarily) with identical position to a previous movement.

I have fixes for these issues which can be reviewed in my branch jalapeno/snapmaker_z_skip in my fork at <https://github.com/jalapeno/PrusaSlicer> .

Treating each of these three issues in turn:

## 1. Incorrect `from` position leads to incorrect movement vector

The branch ms_fix_xyz_travel_speed (SPE-2493) merged at commit ID 6d19776942f0ea059292bd3e310764a1f9ce3d22 changes
* GCodeWriter::travel_to_xyz()
* GCodeWriter::get_travel_to_xyz_gcode()
to remove the `from` argument.

The code is changed to use the internal m_pos member variable for the `from` point.
This partially fixes the first aspect of the Z speed / skipping Z steps problem:
That the movement vector in GCodeWriter::get_travel_to_xyz_gcode() is calculated incorrectly because the m_pos member variable is not updated by code which calls get_travel_to_xyz_gcode() method directly.
There are two places in the code which manually update the from position after calling get_travel_to_xyz_gcode():
* GCodeGenerator::change_layer()
* WipeTowerIntegration::append_tcr()
There are seven places where get_travel_to_xyz_gcode() is called which do NOT update m_pos.
Subsequent calls to get_travel_to_xyz_gcode() will reference incorrect from positions and potentially calculate incorrect speeds.
This issue is addressed in my commit 182a378d81895f690a1fc14733fbc77088b481f8 which REMOVES get_travel_to_xyz_gcode() and replaces it with a new function more closely aligned with what it is actually used for: to force a move without regard to the length of the move distance with respect to EPSILON.

## 2. Z Speed limitation is computed without regard for vector speed decomposition according to the movement unit vector.
This topic has been partially discussed in comments above.
I have added a unit test to check that the Z speed setting is respected in commit c920d590adbca9fdacd58c9022bdb61fa0375bd1. This unit test FAILS and demonstrates that the Z speed limit is NOT respected by the old speed calculation in get_travel_to_xyz_gcode().

Commit 9efbd3a74aae03c2f38e0828694e3565358ef843 corrects the speed calculation so that the Z speed limit is respected. The calculation decomposes the speed into it's cartesian components using the movement unit vector:
distance = to - from
unit_vector = distance / norm(distance)
Then the speed is decomposed into it's cartesian components using the unit vector:
vector_speed = speed * unit_vector
Then the Z speed is checked:
if abs(vector_speed.z()) > speed_z
In this case, the speed is re-computed such that the z component will be equal to speed_z:
speed = speed_z / abs(vector_speed.z())

This ensures that the speed limitation in the Z coordinate is respected.
The unit tests of commit c920d590adbca9fdacd58c9022bdb61fa0375bd1 demonstrate that this works properly.

## 3. EPSILON is inconsistent with XYZF_EXPORT_DIGITS
There are further problems with EPSILON. It is defined inconsistently with XYZF_EXPORT_DIGITS.
The value of EPSILON=1e-4 (0.0001) in libslicer3r.h is inconsistent with XYZF_EXPORT_DIGITS=3 (0.001).
I have addressed this issue by changing relevant uses of EPSILON to XYZ_EPSILON, which is calculated directly from XYZF_EXPORT_DIGITS, thereby forcing it to stay in sync (commit 89ae2c406c06b6dfc07508aaedb3a038219557cc). I have provided unit tests for XYZ_EPSILON in commit 9e2c6e87ae7ed84f993bee1c9881bb01bcffe599.

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1), thanks to @jalapenopuzzle and the PR referenced above. Closing.

closed 2025-02-18 by lukasmatena