prusa3d/PrusaSlicer #13424: Bad G-Code

adiazly, 2024-09-24:
### Description of the bug

Randomly when generating the gcode, it is putting an incorrect code, instead of putting M190, it puts M109 which heats the Hotend and not the bed.
[image]

### Project file & How to reproduce

[3d3_Dummy13_Armor_Parts.zip](https://github.com/user-attachments/files/17114911/3d3_Dummy13_Armor_Parts.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows10Pro

### Printer model

Sapphire Pro

neophyl, 2024-09-24:
The devs are going to need a copy of your saved project file from Prusa Slicer to properly debug this.

I can see from your gcode that your start routine just consists of G28 home and then a mesh levelling.  there are no lines to control extruder or bed heating.  Normally you define those to get full control otherwise PS can\will auto add in commands if the 'Emit Temperature commands automatically' tickbox is enabled.  Something that we cant tell from the gcode if its enabled or not.

Basically a 3mf project file saved from within PS will contain all the profiles and every setting.  I would bet if you had a properly set up start routine as part of the printer profile then it would work reliably.  

Also your gcode flavour is set to Makerware according to your gcode file.  According to this page http://makerbot.wikidot.com/gcode  M109 is
"M109 heated platform temperature (Snnn) (Tn)"  so M109 **IS** heatbed temperature.  M109 is Hotend temperature in Marlin.

adiazly, 2024-09-24:
> The devs are going to need a copy of your saved project file from Prusa Slicer to properly debug this.
>
> I can see from your gcode that your start routine just consists of G28 home and then a mesh [...]

The temperatures are those parameterized in the PrusaSlicer interface, but the generated gcode does it wrong and for the bed temperature it puts M109 instead of M190. It has never been necessary to define temperatures in the start gcode to set bed and hotend temperatures, I think you are wrong.

neophyl, 2024-09-24:
Ok, but as I pointed out your Gcode **flavour** appears to be set to MAKERWARE from your sample gcode and **NOT** Marlin.  
Marlin uses M109 to set the Hotend temperature, while Makerware uses the same M109 code to set the HEATBED temperature.  
If the gcode flavour is set to the wrong type for your machine its would do precisely what you describe your problem as being.

Please attach a saved project file so they issue can be debugged fully.

closed 2024-09-24 by adiazly