prusa3d/PrusaSlicer #13425: travel lift setting generate code problem

pxysea, 2024-09-24:
### Description of the bug

The Retraction length is set to 3.9, and the option to retract at layer changes is enabled. The generated G-code for the first layer slice is as follows:

```
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 E-3.9 F2400
G92 E0
G1 Z.4 F3600
G1 X81.583 Y92.426
G1 Z.2
```
However, the G-code for other layer slices is:

```
;LAYER_CHANGE
;Z:0.4
;HEIGHT:0.2
G1 E1.65761 F2400
G92 E0
G1 Z.4 F3600
G1 X86.861 Y94.913 Z.6
G92 E0
G1 X92.068 Y98.125
M73 P23 R1
G1 Z.4
G1 E3.9 F2400
```
The expected retraction code should be G1 E-3.9 F2400 for other layers as well, but instead it generates G1 E3.9 F2400.
[image]
[image]
[image]

### Project file & How to reproduce

printer -> Retraction length

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.1

### Operating system

Windows 11

### Printer model

DIY Marlin GenL V2.1

rtyr, 2024-09-24:
Please provide your 3MF project.

pxysea, 2024-09-25:
this is  3mf file
[cone_test.zip](https://github.com/user-attachments/files/17122932/cone_test.zip)
I have another idea, which is to have a Z-axis lift and retraction operation before each seam. Is this achievable within the software? Thank you.

rtyr, 2024-09-25:
The g-code output is correct. You are using absolute E axis positioning. You can enable relative E distances parameter in printer settings if you want to change that.

[image]

> I have another idea, which is to have a Z-axis lift and retraction operation before each seam. Is this achievable within the software?

No.

closed 2024-09-25 by rtyr