prusa3d/PrusaSlicer #13426: Slowdown over bridges

IsaiahJoy1, 2024-09-24:
**Is your feature request related to a problem? Please describe.**
When printing PEGT, bridges fail due to the layer placed on top of it. I have to slow down the entire print in order to not brake the bridge.  

**Describe the solution you'd like**
I'd like to see the bridging options that Cura offers, 3 customizable layers of bridging. However, a simpler solution is adding a second layer to bridges or being able to slow the entire layer if the previous contains a bridge.

**Additional context**
Given that not all materials make good bridges, having a way to build up a bridge over multiple layers would extent the capabilities of the printer without sacrificing speed.

Jan-Soustruznik, 2024-09-26:
Hello, @IsaiahJoy1  could you share your project file(3MF), where described problem is located, please?

IsaiahJoy1, 2024-09-26:
[test ponti.3mf.docx](https://github.com/user-attachments/files/17149197/test.ponti.3mf.docx)

IsaiahJoy1, 2024-09-26:
Hello @Jan-Soustruznik , I've up loaded one of the test prints. I can get the bridge do well, but if I want to print the test at a regular speed the bridge gets destroyed by the layer on top of it. I know PETG isn't the best for bridging, but I've gotten grate  results in Cura by utilizing three layer bridging functions to slow down when printing over bridges. At the end off the day, it would be nice to not have to slow down the entire print just because it contains a bridge or have to manually slow it down right above a bridge.

BrickPrototype, 2024-10-04:
Hi @IsaiahJoy1,

Forgive me if you've already tried this or if I misunderstand. 

I thought that the Dynamic Overhang Speed setting was meant to help with cases such as this.

[image]

Jan-Soustruznik, 2025-02-18:
Closed. The slowdown of the first layer of solid infill above bridges (`Print Settings > Speed > Over bridges`) has been implemented in the new release of [2.9.1-Alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1)

closed 2025-02-18 by Jan-Soustruznik