prusa3d/PrusaSlicer #13427: When supports are enabled and "top contact z distance" is set to zero, bridges aren't generated

thatguymendel, 2024-09-24:
### Description of the bug

This is a picture where the supports are generated where i painted and i would expect bridges to generate where there are not supports. This is happening because "top contact z distance" is zero. It would make sense for the  bottom solid infill to look the same as the top solid infill if you are using soluble supports but if I didn't want to put supports on a section and just let it bridge prusa slicer does not generate bridges.

I made this model to represent when you might want to have supports in one area and just let it bridge in another
The left side is flat so it can bridge just fine but the right side is angled and if it tries to bridge one side will not be supported by the model

[image]

This is what happens when "top contact z distance" is not set to zero.

[image]

### Project file & How to reproduce

[Support test.zip](https://github.com/user-attachments/files/17117507/Support.test.zip)

You can test this by changing "top contact z distance" from 0 to 0.1 and you will see the bridges generate how they are supposed to

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 10

### Printer model

Sidewinder X1

neophyl, 2024-09-24:
Correct.  Prusa Slicer is designed to do that.  A contact distance of Zero in effect means that you arent using bridging.  
If you read the tooltip for that setting this is mentioned
[image]

The usual work around if you still want to use bridging elsewhere is to set a very very small z contact distance.  Something like 0.001 or similar for example.  

This has been raised before, for example https://github.com/prusa3d/PrusaSlicer/issues/13298 lists a previous one with links to several more as well as the link to the original work around.

thatguymendel, 2024-09-24:
Thank you, I did not see the tooltip, but I did say it looked like intentional behavior. 

The logic behind doing it this way makes sense if the only reason you would have "top contact z distance" set to zero is because you are using soluble filament and you may as well put supports anywhere it needs it, meaning there is no reason to differentiate between an overhang that is supported and an overhang that is not. A good example of why you might set it to zero without soluble support is if you are using petg for supports while printing in pla. You should be able to print supports from the built plate or where painted and letting harder to reach or harder to remove areas bridge like in #10939. 

The 0.001 trick works well and basically does what an actual fix would do. The difference is that a bridge is also generated where the supports are and as i said previously you would want the bottom solid infill to look the same as the top if its being supported. 

To have both bridges and solid infill where there is supports, they would have to detect if the overhang is being supported before the bridges/solid infill is generated. That may be hard to do and that's why they just blanket turn off bridging.

(the 0.0001 trick worked for what i needed it to. I mostly wrote this just to have some reasons why they might want to change the intended behavior written down)

Jan-Soustruznik, 2024-09-26:
https://github.com/prusa3d/PrusaSlicer/issues/8884

closed 2024-09-26 by Jan-Soustruznik