prusa3d/PrusaSlicer #13429: Pop-up sign "Flow::spacing() produced negative spacing. Did you set some extrusion width too small?" with several files and config

OptimusPrint3d, 2024-09-25:
### Description of the bug

Hello!

On several occasions, I encountered an issue when trying to slice an STL file. I saw on some forums that switching the slicer engine from "Arachne" to "Classic" could solve the issue.

At the time, I tried that, and I was able to slice the file without any problems, but now the same issue occurs using both the Classic and Arachne engines.

The specific problem is the pop-up error: "Flow::spacing() produced negative spacing. Did you set some extrusion width too small?"

Sometimes, removing or adding top layers would allow me to slice, but in other cases, there's no way to fix it, and the same error keeps showing.

My Prusa Slicer version:

PrusaSlicer Version: 2.6.1+win64
Build: PrusaSlicer-2.6.1+win64-202309060711

Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 16,425MB

OpenGL installation
GL version: 4.6.0
Profile: Core
Vendor: ATI Technologies Inc.
Renderer: AMD Radeon(TM) Graphics
GLSL version: 4.60.0

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 10 (build 19045), 64-bit edition

### Printer model

ARTILLERY GENIUS PRO/X2

neophyl, 2024-09-25:
Please attach a 3mf project file when you open a new issue, it makes debugging issues so much more straight forward. 

Are you by any chance moving between different slicers ?  This is commonly seen when opening stuff from something like Super Slicer or similar.  Check your Bridge Flow ratio setting in Print Settings>Advanced>Flow or your Max layer height setting.   For layer height its Normally its something like 75% but we have seen stuff like 75 in there which it interprets as mm. Same with the bridge flow ratio, if you get  95 in there instead of 0.95 it creates issues.

With a saved project we could take a look and see immediately.

OptimusPrint3d, 2024-09-26:
Hello! Thanks for your response, and sorry for the delay.

I have already reviewed the information you gave me, and everything looks good. Right now, I am having the same issue with another file.

I'll prepare the 3mf project for you to take a look at it. Maybe there's something I am overlooking (although I've been using Prusa for a long time, and this is the first time I'm starting to have these issues).

Thanks!

[SAME ERROR WITH OTHER FILE.zip](https://github.com/user-attachments/files/17151578/SAME.ERROR.WITH.OTHER.FILE.zip)

neophyl, 2024-09-26:
Looks to be your Arachne>Minimum perimeter width value causing the issue.  If I change your 25% (which is a extrusion width of 0.1mm) to 26% then it slices without an error.  You wouldn't think it would make such a difference.  Surprised me. 
I double checked with a fixed value of 0.1mm , same error.  0.105mm and no error.

What is really strange though is changing the slicing mode to classic, which *should* make all the Arachne settings irrelevant does not make the error go away.  There must be some check for that parameter that is done even if Arachne is turned off.

That should at least get you slicing again but It's probably best if one of the Dev's can take a look as to why.  Might need some extra input settings validation added to stop it happening.

Jan-Soustruznik, 2024-09-27:
Hello @OptimusPrint3d , Thank you for share the 3mf file. Now we can evoke the issue, so we will try to fix in some next releases. Have a nice day 

Issue created: SPE-2496

@neophyl  , Thank you for your comment. Have a nice day

BloodyRain2k, 2024-12-01:
I have the same error when I slice a specific model, but only when I slice it with more than 2 perimeters.
2 Perimeters is fine, >2 throws the error. But in my case it seems to be Arachne specific, as it goes away when I turn it to "Classic".

But it also does not seem to be the same cause that @neophyl described, as I can enter anything from 0.1 = 0.4 or 50% to 100% and will not change anything.

[fritz_blechv_links.3mf.zip](https://github.com/user-attachments/files/17968700/fritz_blechv_links.3mf.zip)

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena