prusa3d/PrusaSlicer #13435: Incorrect XY compensation.

AlexanderA80, 2024-09-26:
### Description of the bug

When i turning on XY compensation (negative value:  -0.1 or -0.075) i'm getting banding on models that have some bump objects on it's perimeters.

[image]
[image]
![photo_2024-09-26_19-
[image]
59-06](https://github.com/user-attachments/assets/c3bb1b69-460b-4d36-b54e-a8873103b544)

### Project file & How to reproduce

[calibration_cube_comp.zip](https://github.com/user-attachments/files/17153077/calibration_cube_comp.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.1+win64

### Operating system

Windows 10

### Printer model

BTT SKR Mini E3 V3 Marlin Bugfix 2.1.x on Aquila Pro machine

Lukrative525, 2024-10-01:
This looks to me like a hardware issue, as the gcode preview does not show these artifacts. PrusaSlicer is likely not to blame here.
[image]

It looks like the only thing linking this to XY size compensation is the overlapping traces on the internal perimeters.
XY size comp = 0:
[image]

XY size comp = -0.075:
[image]

AlexanderA80, 2024-10-03:
Just checked mechanics of my printer and did not find any problem. I'm using linear rails on X and Y and direct extruder.
I noticed that inregion with letters printing order of perimeters is changed. Outside perimeter of inner wall starts printing first. But in settings Outside perimeters must be print after Inside. Maybe this is the key.
[image]

AlexanderA80, 2024-10-07:
Topic can be closed, i will use flow correction(right cube on pattached photo) instead XY correction.  Bug is still there but i don't know how to fix it, maybe it's a bug of Arachne engine.
[image]

closed 2024-10-07 by AlexanderA80

Lukrative525, 2024-10-08:
I'm not sure where this bug could be. From the images of your prints, I would expect there to be higher extrusion in "perimeter segment 2" relative to "perimeter segment 1," which would result in the swelled extrusions we can see in the middle layers of your print. But the relative error between them is 0.00396 percent, which is, for practical purposes, a non-existent difference. In addition, the y-values of the extrusions are equivalent.

### Perimeter Segment 1
distance travelled = 127.45 - 108.05 = 19.4
extruder advancement = 10.75795 - 10.10128 = 0.65667
extruder advancement per distance travelled = 0.65667 / 19.4 = 0.03384896907216494845360824742268
[image]

### Perimeter Segment 2
distance travelled = 127.45 - 123.276 = 4.174
extruder advancement = 3.36037 - 3.21909 = 0.14128
extruder advancement per distance travelled = 0.14128 / 4.174 = 0.03384762817441303306181121226641
[image]

I went ahead and printed a couple of these. One with your print profile, and one with mine. They both show a tiny bit of what you're seeing, but it's only visible in really unfavorable lighting:

### Regular Lighting
[image]

### Lit From Directly Above
[image]

The one on the left was printed with your print profile (though obviously with my own filament and printer profiles), and on the right is all my own profiles. Both prints honestly seem pretty good to me, so maybe there are some adjustments you can make to your filament profile? Maybe your extruder isn't accurately advancing the filament? Or maybe your y-axis belt is loose? Either way, still doesn't seem like a PrusaSlicer bug to me.

It's a tricky one for sure. Best of luck figuring it out my friend!