prusa3d/PrusaSlicer #13439: Prusa Connect tab is blank even when printer is connected

feesta, 2024-09-27:
### Description of the bug

After a while of having PrusaSlicer open, PrusaConnect tab stops being updated and eventually shows up blank. The upper-right user dropdown shows the logged in account.

[image]

When in this state, PrusaConnect webpage still shows accurate connection to printer. PrusaSlicer can also send to printer from the Plater tab and initiate multiple prints. Restarting PrusaSlicer fixes the issue for a short time. This issue was also present in version 2.8.0.

PrusaSlicer
Version:   2.8.1+MacOS-arm64
Build:     PrusaSlicer-2.8.1+MacOS-arm64-202409181355

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 14.6.1 (Build 23G93)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   4.1.0 (4.1 Metal - 88.1)
Profile:      Core
Vendor:       Apple
Renderer:     Apple M1 Max
GLSL version: 4.10.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>

### Project file & How to reproduce

1. Use PrusaSlicer over several hours or days.
2. Switch between Plater to initiate prints and PrusaConnect to check status. 
3. PrusaConnect will stop updating. 
4. Click Prusa log and the whole tab will go blank.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1+MacOS-arm64

### Operating system

macOS Version 14.6.1 (Build 23G93)

### Printer model

Prusa MK4

ChrisHeerschap, 2024-09-30:
I'm seeing the same, also on Mac OS. The only way to reconnect is to exit the app and come back in. Apparently in windows, pressing CTRL-R or F5 reloads the page and redoes the connection, but this doesn't work on Mac, I've tried every combination of keycodes that I could think of that might trigger a reload.

Jan-Soustruznik, 2024-10-02:
Hello, @feesta , thank you for your report. Could you more specify the point no.4 of your steps, "Click Prusa log ... " what do you mean "Prusa log". Thank you

feesta, 2024-10-03:
@Jan-Soustruznik typo - I meant 'logo'. I updated. I assume that tab is a webpage, I used the Prusa Logo as a refresh to load a fresh page when Ctrl-R didn't work on Mac.

Martymakes, 2024-10-15:
I'm having the same issue. PC + MK3S+

I'm going to try reverting my firmware. Nothing else has worked

sad sad

ChrisHeerschap, 2024-10-16:
> I'm having the same issue. PC + MK3S+
> 
> I'm going to try reverting my firmware. Nothing else has worked
> 
> sad sad

@Martymakes - I've heard that if you refresh the page using CTRL-R or F5 on Windows, that works. I don't have Windows to test it with, but that might help you. None of those work on Mac, unfortunately.

nerakhon, 2024-10-20:
I've tried with cmd+R on mac, (equivalent) but it does not help. Btw the problem seems specific to older macOS releases, on sequoia it works.

ChrisHeerschap, 2024-10-25:
I am running sequoia and it does not work for me. Just re-confirmed. 15.0.1 which is the current version as of this morning.

quintesse, 2024-11-08:
Seems similar to #13137 which was supposedly fixed in 1.8.1, right?

feesta, 2024-11-08:
@quintesse Correct, this seems to be fixed now. PrusaSlicer surfaces token errors and sometimes logs me out allowing me to login to reauthenticate.

Jan-Soustruznik, 2024-11-22:
Hello, @feesta, could you check with latest release 2.9.1-Alpha1, there is some modifications with security tokens.

feesta, 2024-11-24:
@Jan-Soustruznik I'm loving the multi-bed layouts but Connect isn't working after leaving PrusaSlicer open over night. The dark background is different than I saw on 2.8. Restarting the app properly logs into Connect. Let me know if there's anything else helpful for troubleshooting the issue.

[image]

Jan-Soustruznik, 2025-06-23:
Hello @feesta,Thank you for the information. Yes, we are aware of this issue and will try to implement some modifications in one of the upcoming releases. For now, we are collecting data for internal testing. If possible, please send us the logs(if you have some, you can used "--loglevel=3") for a deeper investigation. We kindly ask for your patience. Thank you, and have a nice day!

closed 2026-09-29 by prusaslicer-bot[bot]