prusa3d/PrusaSlicer #13440: G-code preview missing different in PrusaSlicer and G-code Viewer

hackwin, 2024-09-27:
### Description of the bug

I noticed in the 2.8.0 that the gcode preview is missing infill in PrusaSlicer.
The G-code is correct in PrusaSlicer G-code Viewer 2.8.0.
Screenshots attached.
[image]
[image]

### Project file & How to reproduce

Sliced a .3MF file.  Configuration available on request.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.8.0+win64

### Operating system

Windows 10 Pro

### Printer model

Prusa MK4

Jan-Soustruznik, 2024-10-02:
Hello, @hackwin , could you share your 3MF file, please? Thank you, have a nice day

hackwin, 2024-10-02:
Github doesn't allow uploading .3MF files.  Does Prusa have a file upload server?

Jan-Soustruznik, 2024-10-03:
Hello, @hackwin, thank you for quick answer .. But I dont see the issue that you mention above, at the picture is disabled visualisation of Perimeters. For the wall is used perimeters - So is visible empty space inside model. If you change the _Print settings > Layers and perimeters > Perimeters > 1_  then the infill will be generated on the walls, but I dont recomend to print with one perimeter only.

closed 2024-10-04 by Jan-Soustruznik