prusa3d/PrusaSlicer #13442: Wipe Tower is like "vapor ware" Give it life, labels, etc?

noybman, 2024-09-28:
**Is your feature request related to a problem? Please describe.**
YES: If you are completely new to adding an MMU, you are presented with a wipe tower when you open a project. As a beginner, you have no clue what this "cube" is, why it is there, or what it's purpose is. Clicking it, interacting with it, right click, etc ---- you literally get NO information anywhere. How are you supposed to even begin to learn "what next"
Now once you know what it is its not "quick" to make it help you.

**Describe the solution you'd like**
First off, label it. If you click on it, make it an actual object with a label in the side windows. At least now a beginner has a clue.
For beginners and others alike, let us change its shape from the 3D view. Let us touch common properties on the view pane. Give everyone a right click menu?

**Describe how it would work**
Left click: labels the item. It selects it like any other object.
Hover over give a tooltip like popup with basic details.
Right click opens a menu with features like extruder settings for the perimeter, size, preferred position etc.

**Describe alternatives you've considered**
None. As of now, and as a beginner, I was completely dumbfounded. The lack of any indicator on the screen as to what it is seems to be a complete oversight.
Everyone is a beginner once!

u89djt, 2024-09-28:
(from a sympathetic random fellow user) There's info in the handbook: https://help.prusa3d.com/downloads/mmu3/handbook
and there are active forums: https://forum.prusa3d.com/forum/original-prusa-mmu3/

noybman, 2024-09-28:
I understand there is info in the handbook and the forums are lacking for a beginner. Thus my point, until you stumble upon it when reading through the book or other means, you literally have NO CLUE what this "thing" is that you can interact with.

A label when you click on it is the simplest "Start Here" indicator.

Seriously - its an oversight?! We all lose our minds when a "new feature" shows up in our OS that we didn't ask for and have no clue what to call it, how to disable it, or configure it..... right????? (Apple does it, Google does it, Microsoft does it... lol). So at the very least a label would be a start.

But even once you've read the forums/book and understand its purpose and begin to configure it to your individual needs, you still have to position it, then "must" go in to the settings to configure it's size if a "square" wont fit on your bed, (as an example). So why not make the most common things easy access?

I hope my feature request hits home :)
I must be crazy, but a label is a must have.

murk-sy, 2024-10-04:
+1

Wipe tower should be listed in the parts list and should also have an option to be enabled/disabled visible from the very front.

Additionally, (moved from default) position and rotation should be part of the project, not print settings. Any time I slightly touch the wipe tower it shows print settings have been modified, which makes me lose track if something was *actually* changed or if it's just wipe tower changes that are showing up.

All that would make auto-arrange work better in projects where you just have wipe tower enabled anyway(for a multitool setup) but print in a single material.

kurtgluck, 2024-10-23:
Would be useful to be able to cancel the wipe tower mid print as though it's an object.  There are times when the wipe tower falls over, and you are printing in air above where it should be.  Just like an object failing in a multi object print. Now while you thought you wanted the wipe tower, perhaps you can live without it - IF the thing would just stop dumping filament into the sky

noybman, 2026-09-18:
Why do I need to comment on this to keep it open? 

It hasn't been addressed, so here is a comment

murk-sy, 2026-09-24:
Wipe tower now listed in objects list in 3.0.0-alpha12

[image]
[image]