prusa3d/PrusaSlicer #13444: Successful slice with wipe tower <> model collision

ninjazamo, 2024-09-30:
### Description of the bug

When the wipe tower does not have a collision in the first few layers, slicing completes. However, if there is a collision higher in the model, no error is generated. 

### Project file & How to reproduce

layers 76-126 including 3mf file with sample collision and screenshot of sample area. 

[ghostwalkingdog_towererror.3mf.zip](https://github.com/user-attachments/files/17182341/ghostwalkingdog_towererror.3mf.zip)
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0+macOS-arm54

### Operating system

macOS 15.0

### Printer model

MK4S with MMU3

Jan-Soustruznik, 2024-10-02:
Hello, @ninjazamo , Thank you for your report. We can confirm that the colision doesnt work correctly. We created internal issue - SPE-2505. Thank you, have a nice day

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.