prusa3d/PrusaSlicer #13445: Urgent--Prusaslicer 2.8.1 broke the functionality of "don't cross perimeters"

sbierly, 2024-09-30:
### Description of the bug

Recently upgraded to 2.8.1, printed production parts as usual but resliced in 2.8.1, noticed significant stringing within the cavities as  it printed upwards with internal perimeters. Really bad. I reverted to 2.8.0, NO OTHER CHANGES, and the problem went away, back to how it was before with nearly zero internal stringing. 2.8.1 update clearly fixed something else, and broke this. It probably went unnoticed since it only really manifests in TPU/TPE (I'm printing at 85A). 

### Project file & How to reproduce

My parts are proprietary, but all you need to do is print something like a cube or cylinder carving out another smaller cube or cylinder in the middle, (be sure to have enough thickness for internal/external perimeters, and some spiral fill) you should get the same stringing even with "don't cross perimeters" turned on. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.1

### Operating system

Windows 11

### Printer model

Mini

rtyr, 2024-09-30:
Reproduced. Thank you.

Orafilynie, 2024-10-01:
I have the same problem.

My STL is just donut shaped cylinder

I noticed that when i activate the functionality, with the seam set as nearest, it "kinda works" (with kinda i mean it doesnt do any curved paths and just straight ones)

And it completly breaks when i use the aligned seam parameter.

## Images :

Unactivated feature :

[image]

Activated feature with nearest seam :

[image]

Weird avoid pathes with nearest seam:

[image]

Activated feature with aligned seam :

[image]

Weird (broken) avoid pathes with aligned seam:

[image]

sbierly, 2024-10-01:
Thanks for adding to this--sorry, yes, I'm using aligned seams as well.

Jan-Soustruznik, 2024-10-02:
created issue: SPE-2498

MongooseTNM, 2024-11-07:
It also causes travel moves to be placed outside the build plate and causes the print to fail I have verified that this only happens when avoid crossing perimeters is set to on and set to 0mm when I set it to 200mm it works fine but when I set it to 400mm the behavior comes back
[image]

sbierly, 2024-11-07:
Wow, it's been a month now, and I'm still  having to use the old software
version.

I'll add one more thing here, although you may tell me it goes elsewhere,
but even in the previous version, there is a related perimeter-crossing
issue that greatly affects TPE printing, maybe in general, at least
printing separate objects. When moving to a new part, Prusaslicer blindly
moves to the center coordinates of the object, descends, and then has to
move sharply out to the first perimeter to begin drawing it. In the
process, it oozes filament right in the middle of the part, often leaving a
little wiggly strand that becomes a scar on the bottom of the part. This is
even if "don't cross perimeters" is set! Kind of a violation right out of
the box, and is really making my parts ugly, and I see no clear solution to
this. Prusa, I hope we can please make sure the behavior matches the
settings.

Thanks,
--Scott

marsmanp, 2024-11-20:
so that would confirm why I can't get these travels to stop crossing.  Frustrating to say the least when trying to print a hollow cylinder and getting terrible spots because of the travels.

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.